grbl/grbl #1009: M3 M4 M5 commands

meatom, 2016-06-07:
Hello,
I've been using my mini laser engraver with GRBL 0.9i, so decided to upgrade to version 0.9j.
The issue is that now the engraver does not turn the laser on with the M3/M03 command. To turn on the laser now I have to use M4, and M3 to turn off. The M5 command doesn't make any effect.
I've measured the voltage at pin 13, that is hooked with an mosfet, and it just goes high with M4 command.
The user  PCPayne openned the issue  #949 with the same problem, but no solution was provided.
I tried to downgrade to version 0.9i but it still not working. Disabling/enabling the #DEFINE VARIABLE_SPINDLE and #DEFINE USE_SPINDLE_DIR_AS_ENABLE_PIN settings in config.h had no effect.
I tried with a different Arduino and measured the pin 13 but it doesn't goes high with M3 command.

chamnit, 2016-06-07:
@meatom : Pin 13 is the spindle direction pin. Not enable. 

Recently, Grbl changed how spindle enable works to be compliant to the LinuxCNC g-code standard. You'll need to set a spindle speed greater than zero and have M3/4 active to ensure that the spindle turns on. So if you have VARIABLE_SPINDLE and USE_SPINDLE_DIR_AS_ENABLE_PIN enabled, pin 13 will not turn on if spindle speed is zero.

If you have VARIABLE_SPINDLE disabled, the spindle enable pin moves from pin D11 to D12 and works as a simple on/off pin. In this case, it doesn't require a S spindle speed word to turn on.

closed 2016-06-07 by chamnit

kardinal7, 2016-07-25:
I'm trying to put GRBL to control the spindle, and I kinda achieve that, but I can't stop the spindle. I can change the spindle speed s500/s100 commands but the M5 command doesn't work. is there any configuration to config the pin value on the stop situation?

Or else what is the pin value in stop situation.

chamnit, 2016-07-25:
@kardinal7 : Make sure you have your pin wired correctly. In v0.9, the spindle enable pin is on D12.

truvisual, 2018-02-27:
@chamnit :I've GRBL 1.1f on a Arduino Mega2560 + Ramps 1.4 and running LaserGRBL.
Can you help me please? It's about PWM. M3/M4 and M5 command work, in that way, LaserGRBL says ok.
But only on and off is possible. The S command doesn't work. I've read alot in the Internet about that.
I'm a absolute beginner in GRBL and Arduino.
The situation, when i type in M3/M4 or M5 it turns on and off. Same by s127 M3 or M3 s127...
But with full power. The PWM is on Servo Pin 4. 

I got a 2,5W blue Laser with a Laserdriverboard (TTL) and 5V.
So, i can't connect it to D9 or D8, because D9 and D8 are 12V.

In LaserGRBL M3/M4 and M5 are for the Laser, in my case, M3 etc. is to Servo Pin D4.
Where do i have to make any changes, that the Spindlespeed is taken effect of the commants S0, S127, S255 etc.?  
Which are the keywords for the reason?
 SPINDLE_ENABLE or  SPINDLE_PWM_???
I got the Lasers TTL on PIN D4, + on D4 and - on ground.
I put a multimeter on it, to messure the V. It's always 5V

Here is a part of the cpu_map.h
```
 // Define spindle enable and spindle direction output pins.
  #define SPINDLE_ENABLE_DDR      DDRG
  #define SPINDLE_ENABLE_PORT     PORTG
  #define SPINDLE_ENABLE_BIT      5 //  5 MEGA2560 Digital Pin 4 - Ramps 1.4 Servo 4 Signal pin
  #define SPINDLE_DIRECTION_DDR   DDRE
  #define SPINDLE_DIRECTION_PORT  PORTE
  #define SPINDLE_DIRECTION_BIT   3 //  3 MEGA2560 Digital Pin 5 - Ramps 1.4 Servo 3 Signal pin

  // Define flood and mist coolant enable output pins.
  #define COOLANT_FLOOD_DDR   DDRB
  #define COOLANT_FLOOD_PORT  PORTB
  #define COOLANT_FLOOD_BIT   4 // MEGA2560 Digital Pin 10 - Ramps 1.4 12v output
  #define COOLANT_MIST_DDR    DDRH
  #define COOLANT_MIST_PORT   PORTH
  #define COOLANT_MIST_BIT    6 // MEGA2560 Digital Pin 9 - Ramps 1.4 12v output

  // Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
  // NOTE: All CONTROLs pins must be on the same port and not on a port with other input pins (limits).
  #define CONTROL_DDR       DDRK
  #define CONTROL_PIN       PINK
  #define CONTROL_PORT      PORTK
  #define CONTROL_RESET_BIT         1  // Pin A9 - RAMPS Aux 2 Port
  #define CONTROL_FEED_HOLD_BIT     2  // Pin A10 - RAMPS Aux 2 Port
  #define CONTROL_CYCLE_START_BIT   3  // Pin A11 - RAMPS Aux 2 Port
  #define CONTROL_SAFETY_DOOR_BIT   4  // Pin A12 - RAMPS Aux 2 Port
  #define CONTROL_INT       PCIE2  // Pin change interrupt enable pin
  #define CONTROL_INT_vect  PCINT2_vect
  #define CONTROL_PCMSK     PCMSK2 // Pin change interrupt register
  #define CONTROL_MASK      ((1<<CONTROL_RESET_BIT)|(1<<CONTROL_FEED_HOLD_BIT)|(1<<CONTROL_CYCLE_START_BIT)|(1<<CONTROL_SAFETY_DOOR_BIT))

  // Define probe switch input pin.
  #define PROBE_DDR       DDRK
  #define PROBE_PIN       PINK
  #define PROBE_PORT      PORTK
  #define PROBE_BIT       7  // MEGA2560 Analog Pin 15
  #define PROBE_MASK      (1<<PROBE_BIT)

  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // 1024.0 Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4C  //  OCR4C
  #define SPINDLE_COMB_BIT          COM4C1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH  // PORTH
  #define SPINDLE_PWM_BIT   4 //  5 MEGA2560 Digital Pin 8

#endif
```

danielgimeno, 2020-09-11:
Hi there, I'm starting with commands M3 M4 and M5, with a RAMPS 1.5 on an Arduino Mega. My question is how can I select from gcode, the different ports (there are 4) in ramps. IS there a kind of "M3 port 1"?

Thank you Daniel.