grbl/grbl #1013: Parsing uninportant issue

vlachoudis, 2016-06-12:
By mistake on the I've typed
```g0 x150 100```
intended to be ```y100```
but then the machine starting moving like it interpreted the " 100" as feed but it was much less than f100.
I know I should have been careful, but I just wanted to mention it :)

chamnit, 2016-06-12:
@vlachoudis : Thanks for the bug report. I'll look into and issue a fix for this right away.

chamnit, 2016-06-14:
@vlachoudis : I couldn't find any reason why the g-code parser would interpret that command as F100. The pre-parser should remove all spaces from the line, so it should be interpreted as a ```g0x150100``` command. Could you check if you get the exact same behavior if you send a ```g0x150100``` on your system? 

It's possible that this very large x value (150 meters) could have caused a numerical problem in the planner, when converting it to steps. The math in Grbl assumes realistic distances don't exceed 10 meters or so, but doesn't actively check for value ranges. I tend to omit these because lots of checks eat up flash quickly and can begin to degrade overall performance. 

If this is indeed a large value problem, I can hold off on creating a fix until v1.0 goes to master. Re-evaluate then to see how we are doing with flash and performance after people have beta tested everything. I'd like to keep as much remaining flash available to me during beta testing so I can have some flexibility to fix any major problem (fortunately I don't anticipate any).

vlachoudis, 2016-06-15:
After the command the gantry was moving with a very small speed, so I thought that it was interpreted as a small feed rate. The funny thing was that Grbl was unresponsive ot possible to feedhold with software or hardware. The only solution was to do hardware reset.

chamnit, 2016-06-15:
@vlachoudis : Thanks. So that seems to validate my thinking that its a large value problem causing a system failure. I've already added a note to fix this in my bug tracker.

vlachoudis, 2016-06-15:
Ok thanks