grbl/grbl #1015: drawing paper with Grbl

jajiyiko, 2016-06-12:
Hi;
I have a usbcnc grbl shield with a4988 drivers. I made simple machine x and y (two stepper motors) axis.,
For example I want to draw a star on paper.But not good.

like following picture;
this shape i want
[image],
this is it is written
[image]

sorry for my english

thanks

milezdr, 2016-06-13:
What CAM program you use for generating the g code? or can you post the g code

jajiyiko, 2016-06-13:
hi!
I use LX Easy G-code Tools software for gcode.
nc file is attached.
[star.zip](https://github.com/grbl/grbl/files/312480/star.zip)
thank you

109JB, 2016-06-15:
Your zip file won't open. At least not for me.

closed 2016-09-03 by chamnit