grbl/grbl #1016: Positive space ?

LETARTARE, 2016-06-13:
To increase the versatility of Grbl , I propose to use either a **negative** or **positive** space as needed., without using "G10 L2(0) Pn ........".

I used  version ' Grbl - 1.0c (edge) svn1837 ' .
Here's the fix:
[grbl-1837-positive.zip](https://github.com/grbl/grbl/files/311539/grbl-1837-positive.zip)
To use the positive space , simply uncomment in ' grbl.h ' line  '#define POSITIVE_SPACE'.

I realized the tests of  homing  on one machine to three Cartesian axes.
The positive direction of the axes movements respect the meaning defined in Linux. 
Sensors which detect the reference position (home) are placed at maximum axis movement.

Example :
dX=500, dY=400, dZ=90, dR=2 mm

A- if  positive space, 

1. with $23=0 -> machine position = 498/398/88 mm
2. with $23=7 -> machine position = 2/2/2 mm

B- if  negative space, 

1. with $23=0 -> machine position = -498/-398/-88 mm
2. with $23=7 -> machine position = -2/-2/-2 mm

I use a personal GUI.

Best regards.

chamnit, 2016-06-13:
@LETARTARE : I get this question once every few months. Because of this, I wrote up a answer in the [Grbl FAQ](https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-or-it-so-annoying-and-not-what-im-used-to) to avoid having to repeat myself over and over again. In short, no, because it really doesn't matter where the machine origin is set since all tool paths are in work coordinates, and there is a compile-time option to set the origin wherever the homing cycle ends up to address this.

closed 2016-06-13 by chamnit

LETARTARE, 2016-06-13:
Yet I am an avid reader , but I had not read this response!

chamnit, 2016-06-13:
@LETARTARE : It's come up on Github several times, but more recently, answering this question has been on other forums, like the ShapeOko, Inventables, and OpenBuilds forums.