grbl/grbl #1018: Speed & frequency

mihaisalajan, 2016-06-16:
Hi everyone

I have a small issue. please help me.

I just start using GRBL V0.8 firmware, my board is arduino uno the stepper is a nema14 200 steps /revolution the driver is a4988, i am turning a screw with 8mm lead that means 25 steps/ mm
but the motor is making a lot of noise and the speed is slow, if i set the steps at 100 steps/mm the axis is moving just fine without noise and the speed is just right but than when i tell it to move 1 mm it moves 4 mm (!!!obvious!!! ) i tried different speeds ($4 adn $5 from 100-3600) but nothing changes. i need to mention that i am using grbl controller 3.6.1 
i also tried different step pulses ($3) from 3 to 30 still without result 
another question how ca i set the tA and tB from picture below at the value of 1µs , I had the impression that this is setup with the speed adjustment

or is it that tA is $3 command and tB is $4and $5 commands ?

step a4988

PS please excuse my english is not my native language

gerritv, 2016-06-16:
First recommend that you move to 0.9j as it has many enhancements over 0.8.
This article describes how to set up your driver: https://www.pololu.com/product/1182 
Often a combination of current limiting and microstepping cause the symptoms you describe.

electrokean, 2016-06-16:
It sounds like you're in the motors resonance zone.  You should switch to microstepping as suggested by @gerritv and then increase your steps per mm appropriately.
You can then increase your max velocity to improve speed to where you want it, although you may also need to tweak acceleration up or down to get the best setting to avoid resonance on the low end, and stalling on the high end.
Pulse widths shouldn't matter - they just need to be long enough for the driver to see the pulses reliably.  Some opto-isolators are slow and need longer pulses.  A low pulse rate will limit your maximum velocity, but is otherwise not a problem.

mihaisalajan, 2016-06-16:
Thank you all for your opinion and suggestions, but i found  the problem by chance.

I had to reset the arduino,  any setup in $4 or $5 it doesn't have immediate effect, i need to  restart arduino again restart connection between arduino and GRBL controller and surprise the speed was adjusted, and with some small calculation I found my 1µs impuls that is the speed of 1200 mm/min 
so every thing is perfect now 

@gerritv I will do it today thank you 
@electrokean i tried it already by setting the pulse/mm at 100 not 25 and the step at 1/4 but the result was the same as before the distance was correct but without improving the speed or escape the noise 

any way now is working fine again, thank you all

closed 2016-06-16 by chamnit