grbl/grbl #1019: Motors has stopped working...

alihusnain2229, 2016-06-16:
Hy there. very first time to post my query in some blog... Actually today I made first test run of my arduino CNC shield V 3.00 with A4998 drivers using 1.8Amp steppers. Things were quite fine in beginning. But then my motors suddenly stopped working. I have check each driver individually, all are working on bread board with some Random sketch. Then i pasted back on the CNC shield. Still no motor showed any response, i can easily rotate them from there shaft like they are not connected to any driver. is some thing has happened to my CNC shield or is it some thing else? 

more ever my GRBL started giving alarm lock error, then setting disabled error on running homing cycle. Why this all is happening to me in the very beginning :'(  

Please help me out..

closed 2016-09-03 by chamnit

Encarnacao, 2016-09-04:
did you adjust the trim pot on the drivers, maybe if you using microsteps on the cnc shield you will need more current to drive the motors, also dont forget if you using grbl 9 on the cnc shield v3 you will need to swap the pins D11 with D12 sense there was some changes, the alarm lock should be cause you set the hard limits on?

gerritv, 2016-09-04:
You might get better advice from a forum that supports you CNC shield? This does not sound like a Grbl problem.