grbl/grbl #1020: State error: Invalid gcode ID:36

sanotronics, 2016-06-17:
Hello!

I am using grbl 0.9g with a Shapeoko 3, and bCNC to control it.

I have succesfully milled, drilled and engraved PCBs.

Now, I am trying to mill a PCB, with a rectangular shape, but with 5mm arcs on the corners.

I am getting the error shown in the image, and the process stops close to the middle of the first arc.

I cannot find any info or anyone in any forum who had a similar problem :S 

I attach the image showing the error and the last position of the mill.

The line I get the error at is this one:

`_**G02** X-138.0000 Y5.0000 R5.0000 F504.00_`

And when I look at the application's terminal, the last commands are:

```
 G1X-133Y0Z-1.682966
ok
 **G1**X-134.950175Y0.396Z-1.670572R5.0000F504.00
ok
 G1X-135.1081Y0.504275Z-1.668173
ok
ok
 G1X-136.591443Y1.521274Z-1.645236
ok
 G1X-137.468275Y2.8919Z-1.616967
error: Invalid gcode ID:36
ok
ok

```

See how the G02 command gets modified by a G1 command?
Maybe by getting a G1 command with the R5.000 is what grbl doesn't like?

Is this maybe a bCNC issue? I will try driving it with a different software and update.

[image]

[Potes4x2.top.mill.txt](https://github.com/grbl/grbl/files/321408/Potes4x2.top.mill.txt)

chamnit, 2016-06-17:
@sanotronics : I'm assuming that since you are milling a PCB, you are also doing surface probing to modify the g-code inside bCNC. If so, bCNC either mistakenly changed the G2 to a G1 command or didn't remove the R word. I would post this issue there. The author is very responsive and usually quick to fix things.

closed 2016-06-17 by chamnit

sanotronics, 2016-06-17:
Yes indeed! it was bCNC's autoleveler, it changed G02 to G01 and tried to "emulate" the arc doing lines :/

It'd be awesome if it gets fixed! 

Can you please copy the link of that issue here please?

sanotronics, 2016-06-18:
Ok never  mind, I thought you meant YOU'd post it, but it  was because I speak spanish as a mother tongue, and I just missed it.

I posted the issue here:
 https://github.com/vlachoudis/bCNC/issues/362