grbl/grbl #1021: Install error.

andsetinn, 2016-06-21:
I have an interesting install error. I am probably doing something wrong or overlooking simple setting.
Anyway, here is my problem. 

I have chinese clone of Arduino mega 2560 that I bought in kit with CNC shield Ramps v1.4, 5xStepSticks and a display. 
I am using Imac with Yosemite 10.10.5 and Arduino IDE 1.6.9
After figuring out that the clone has the CH340 USB chip I finally got it talking to the Mac.
I can download the Blink program to the Arduino and the LED will blink happily. So the Mac, Arduino IDE and the Arduino are working, as such.
The GRBL is more difficult. 
I set up the library according to instructions, grblUpload compiles and uploads (uploads in 20 seconds) without errors. 
But the Serial Monitor displays blank screen even with "115200" baud and "Both NL & CR" selected.
When I start the Universal Gcode Sender it claims that the port is busy until I turn off the Arduino IDE. The it opens the serial port and when I go to "machine control" and select manual operation it claims that "GRBL has not finished booting"

I have tried multiple restarts of Arduino IDE and the clone. My conclusion is that the grbl-master folder (created Thursday 17 March 2016 17:54. Zip file is 377KB) is corrupt or some settings in there are wrong for my setup. 

Does anyone have suggestions?

chamnit, 2016-06-21:
@andsetinn : First, Grbl is not compatible with RAMPS. Second, CH340 USB-serial chips have known communication problems, which often result in lost data. (It doesn't explain your connectivity issue). Third, make sure you are altering the config.h for Mega2560. Fourth, the Mega2560 is not officially supported by Grbl at this time, so you may run into problems. However, it should be working fine. Lastly, when nothing seems to work, disconnect everything from the Arduino and start simple.

closed 2016-06-21 by chamnit

andsetinn, 2016-06-21:
Thank you Chamnit. 
At the moment I don't have the RAMPS shield on the Arduino.  
I will look at config.h and see if that helps me with my problem.

DrZ1968, 2016-06-21:
@Andsetinn try the marlinkimbra repository

andsetinn, 2016-06-21:
Thank you both for the help.
I tried old GRBL from EliteEng and now I get something on the Serial monitor. It´s an error but a step in the right direction. :) I will look at malinkimbra and others and try to figure out how to get this to work.

DrZ1968, 2016-06-21:
What are you wanting to use it for, laser?  3d printer? Mill?

andsetinn, 2016-06-21:
At the moment I'm thinking of mill with the option of a laser, if I can buy one. First step is to get the electronics working. Then I have to design my own mill since I'm not fond of the most popular ones available at affordable price. :) 
By the way, I programed an old Duemilanove and GBRL seems to work on it so the files I have are fine. Now I need to wait for smaller motorshield (and UNO) to arrive in the mail.