grbl/grbl #1022: The problem with the probing

maximius, 2016-06-22:
First of all I want to say thank you for the much needed development. Prior to that, worked with the program mach at lpt, but it is very inconvenient. Now moved his CNC router to Arduino Uno + cnc shield v3 + grbl 0.9 j.

For milling printed circuit boards found a very useful program GrblHeightProbe, which takes into account the surface roughness when milling. But when you work a problem using this program. After the beginning of the probing PCB after the command g38.2 grbl replied "Probe alarm" and stopped responding to other commands. The work can be continued only after the restart Uno. The same error appears when after the command G38.2 I raise the spindle G91 command, and then want to run the command G38.2. What could be the problem? Please help in this issue.
[image]
[image]

jaketeater, 2016-06-22:
With G38.2, if the probe does not contact anything, it gives an alarm.  With G38.3, it will not give an alarm if the probe fails to contact, but I believe it will give you a PRB result, but with the max distance you specified.  

In your case, it looks like the Z dropped to -5, but failed to contact the PCB.

chamnit, 2016-06-22:
@maximius : You need to insert a motion to move the probe off the target. Grbl checks the state of the probe prior to executing a G38.x. If it detects that its triggered when its not supposed to, then it'll alarm out.

closed 2016-06-22 by chamnit