grbl/grbl #1024: Grbl reponding with two 'ok'

juliolarab, 2016-06-25:
I'm new to grbl, build a system and set up with a raspberry, arduino, and command it with ugs.
Every thing seema to work well.
Then i decide to get more in depth, and base on the simplestream code, y write the following (also new to python ) and found that grbl is respondion with a double 'ok', as can be seen change the readed value to discard a double printin of the same read.
This is normal  ?, in ugs there is only one 'ok' respose  or there is i'm doing wrong,
I understand the strean by counting method philosophy, and suppose that for that working  as you recieve the 'ok' for each command  you can update your ques, but i belive there be only one respose per command. I' right?

this is thw code

#!/usr/bin/env python
import serial
import time

def OpenSerial():
    s = serial.Serial('COM4',115200,timeout=0.1)
    return s

def CloseSerial(s):
    s.close() 

def ReadLines(s):
    time.sleep(0.1)
    while s.inWaiting():
        grbl_out = s.readline() 
        print grbl_out.strip()
        grbl_out='-'
        print grbl_out.strip()
       
Ard=OpenSerial()
Ard.write("\r\n\r\n")
time.sleep(2)
ReadLines(Ard)
Ard.flushInput()
while True: 
    cmd=raw_input("  Enter  grbl command.").upper()
    if "END" in cmd:
        break
    print '  Sending: ' + cmd    
    if "^X" in cmd:
        cmd=b"\030\r\n"
        Ard.write(cmd)       
    else:
        Ard.write(cmd + '\r\n')        
    ReadLines(Ard)                       
CloseSerial(Ard)

And this is the result

-
Grbl 0.9j ['$' for help]
-
['$H'|'$X' to unlock]
-
  Enter  grbl command.$x
  Sending: $X
[Caution: Unlocked]
-
ok
-
ok
-
  Enter  grbl command.?
  Sending: ?
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
-
ok
-
ok
-
  Enter  grbl command.g0x100
  Sending: G0X100
ok
-
ok
-
  Enter  grbl command.g0y100
  Sending: G0Y100
ok
-
ok
-
  Enter  grbl command.?
  Sending: ?
<Idle,MPos:100.000,100.000,0.000,WPos:100.000,100.000,0.000>
-
ok
-
ok
-
  Enter  grbl command.end
>>>

109JB, 2016-06-25:
When writing to GRBL you only need either a \r or a \n.  By sending \r\n grbl is interpreting this as 2 lines sent.

juliolarab, 2016-06-25:
as simple as that., sorry.

Thanks

closed 2016-07-02 by juliolarab