grbl/grbl #1028: HELP! 2 Motors to 1 Axis!

kennet67, 2016-06-28:
This is my situation:
i have 2 Stepper Motors Nema 17 And a steppermotor driver a4988 pololu.
So I tried to put 12V to the driver and the nema 17 doesnt have so much strength I dont know Why the VRef its 0.60 iF you ask.
And I have a threaded rod so when i put only 1 motor doesnt move my structure but i want to ask.. How i can put the another one in mirror mode... I mean Example The motor turns right And the another Have to turns left... I tried to put a 74ls04 to deny the output to step and dir... But sometimes works and sometimes No... Im using Gcodesender and an arduino UNO and grbl 0.8... Please Help

chamnit, 2016-06-28:
@kennet67 : You have two options. Wire your steppers to the same driver with one of the motor windings leads switch to make it run the other direction. Or use two stepper drivers and place an inverter on the direction pin on one of them.

closed 2016-06-28 by chamnit

langwadt, 2016-06-28:
@chamnit with two stepper drivers you can also just swap one of the windings and skip the inverter ;)

chamnit, 2016-06-28:
@langwadt : You're totally right. That's an example of an utter failure of drinking enough coffee. Time to brew a new pot.

kennet67, 2016-06-28:
@langwadt I didnt get it, Wich thin gi have to swap? Sorry my ignorance

jahnj0584, 2016-06-28:
If the first motors cables are ab cd in the terminal, motor two goes cd ab

lalo-uy, 2016-06-29:
I belive you have to  change from abcd to abdc

kennet67, 2016-06-29:
the pololu goes from 
------> 1b, 1a, 2a, 2b
------> green, black, red, blue

so i have to put it in this order?

----->1b, 1a, 2a, 2b
------>blue, red, black, green?
??? 
and i have to use a NOT inverter? 7404 to deny the Direction pin? and its only the dir pin? leave the step pin normally?

thnaks ofr the help

chamnit, 2016-06-29:
@kennet67 : Just make sure you are connecting the coils correctly. It's easy to know if two motor wires belong to the same coil by checking for near-zero resistance between them. The wire numbers and letters often swap between motor spec sheets and stepper drivers, so don't trust them. As long as you connect the two ends of a motor coil to where the driver wiring diagram expects a coil, you're good. If you want the stepper to switch directions, swap the two ends of one coil on the driver (NOT BETWEEN COILS). That's it. Wire everything else normally into Grbl.

kennet67, 2016-06-29:
@chamnit i got it Thanks but i have another problem... the nema 17 motor doesnt have strength it has a minimal strength im using a super heavy duty power supply 17V and 10 amps but stills the same =( 
still using the a4988 driver (i'm not using a shield) ...how to fix it =(

chamnit, 2016-06-29:
@kennet67 : Increase motor driver voltage as high as you can. This increases stepper motor torque. Otherwise get a Nema23.

kennet67, 2016-06-29:
@chamnit sorry for bother you but i have a nema 17 42BYGH47-401A i bought it from here:

https://www.amazon.com/Nema17-Stepper-Motor-4-lead-Connector/dp/B00QEXSCE8?ie=UTF8&keywords=nema17&qid=1465484715&ref_=sr_1_4&sr=8-4 

and How much VREF do i have to regulate? i have 0.60V BUT STILLS CRAP! =( No strength

lalo-uy, 2016-06-29:
Are you sure the current liimit in the a4988 is set right?

kennet67, 2016-06-29:
@lalo-uy How to know if it's right? the voltaje between the screw and the gnd its 0.60V now but wich volatge i need to regulate it?

lalo-uy, 2016-06-29:
Check this  http://reprap.org/wiki/Pololu_stepper_driver_board

But also check the value of the Rs in your board to be sure.

Nema 17 shuld handle 1 to 1.2 A

pth3k, 2016-06-29:
0.6v comes to 1.5a / coil using the formula from the riprap wiki  I_TripMax= Vref/(8*Rs) 

Rs = Sense Resistor on Pololu = 0.05 ohms

Amps = 0.6 / ( 8 x 0.05 ) = 1.5A

You probably want to set Vref to 0.8v to get the 2A MAX of your steppers

Be warned, they will probably be pretty hot at that current.

You probably should move up to NEMA 23’s. You can double or triple the torque for the same amount of current, 
depending on the brand/model of the NEMA 23.

Ty Hoeffer

jahnj0584, 2016-06-29:
nema 17s arent good for much besides light weight 3D printers.