grbl/grbl #1032: Gcode not executing corectly??

jureledinek, 2016-07-01:
Hello CNC enthusiasts,

I have a problem with GRBL. I am trying to cut out a sign made with Inkscape. I generated the gcode using MakerCam. 
Here is the layout in Makercam:
[image]

Gcode in simulator (CAMOTICS):
[image]

When I send the Gcode to the machine, the machine moves very strange, like a zig-zag movement where should be a straight line. Here is an example, I draw the movement of the machine with yellow.
[image]

Here is first couple lines of code:
(Generated by PartKam Version 0.05)

G21 G90 G40

(pocket 1)
G0 Z3
T0 M6
G17
M3
G0 X100.98984771573605 Y139.46522842639595
G1 Z-1 F300
G3 X102.26395939086295 Y139.91370558375635 I-95 J272.23350253807104 F500
G3 X103.07360406091371 Y140.2030456852792 I-38.7741116751269 J109.45939086294416
G3 X103.497461928934 Y140.35786802030455 I-21.619289340101524 J59.62436548223351
G3 X103.73096446700508 Y140.44670050761422 I-6.375634517766498 J17.200507614213198
G3 X103.86040609137055 Y140.497461928934 I-3.5939086294416245 J9.302030456852792
G3 X103.95431472081219 Y140.53807106598984 I-1.3172588832487309 J3.2741116751269033
G3 X104.13705583756345 Y140.62944162436548 I-0.817258883248731 J1.8730964467005076
G3 X104.90355329949239 Y141.93908629441626 I-0.7335025380710659 J1.3096446700507614
G3 X103.94670050761421 Y143.33756345177665 I-1.50253807106599 J0

I tried sending commands manually one after another (I'm using Universal Gcode sender) and the movement of the machine was the same.

Does anybody have any idea what could be the problem? 

Thank you a lot and best regards,
Jure

109JB, 2016-07-01:
First thing I would change is the number of decimal places. You only need a max of 3 or 4 decimal precision. Each line of your code is about 80 characters in length.  I can't recall off hand what the line length limit for grbl is but I think it is less than what you have.

Second, all of your code is for G3 counterclockwise arc movements, which makes little sense to me.

jahnj0584, 2016-07-01:
Also if you have an Xand a Y command it won't be a straight line. Sounds
like you have the wrong post settings and don't have it set up for GRBL
(should be a .nc or txt file

gerritv, 2016-07-01:
Or the pocket (zigzag) operation is on the outside of the shape, not the inside?

closed 2016-07-11 by jureledinek