grbl/grbl #1034: Making the Fan switchable on Stepoko card

subdude27196, 2016-07-02:
I have a Stepoko controller board running on a Shapeoko 3 CNC.  The board has an adequate heat sink, so the fan itself is not included on the card.  However, the 12V Fan Switch is still installed, and can be used to power other 12V devices.  Unfortunately, the fan is unswitched, so it can't be turned on/off via GRBL commands.  Is there a way to modify this?  i'm seeing a list of G-Code commands that includes M106 - (Rep/Rap) Fan On and M107 - (RepRap) Fan Off.  Isn't there a way to modify/add code to the grbl Master files to enable the GRBL commands to include switch the Fan on and off?  (NOTE: I'm definitely not a programmer, so any suggestions would need 'GRBL for Dummies' step-by-step instructions!)

[image]

chamnit, 2016-07-02:
@subdude27196 : I'm not sure how the StepOko board's fan is wired, but I think it's hard-wired into the input voltage (most are). So you won't be able to control how or when the fan power is running, other than when the system is powered up.

FYI, Grbl is specifically for subtractive CNC, not 3d printers. 3d printers have their own set of g-code that has grown over time and has since become a distant dialect of g-code to what's considered the standard. Grbl only contains a base set of g-code that is standard among milling machines (even then it's still a dialect due to slight implementation differences among manufacturers). The option of controlling a fan isn't among them, as it's not often needed.

That said, Grbl is restrictive to keep things clean and simple for the time being. Opening up and adding more supported g-code in the future is definitely in the cards, but not in this implementation of Grbl. It will have to be in the upcoming ARM version.

closed 2016-07-02 by chamnit

subdude27196, 2016-07-02:
Thanks for the info.  I just discovered that RepRap is a brand name of a 3D printer, so I can see now that they have implemented a G-Code command on their card to turn their fan off and on... but it only works with GRBL; *ONLY* with the RepRap controller card.

There are 5V and 12V pins showing on this card, and it's an open source card, so I should be able to source those to run other devices.  The question is how.....and can they be controlled with G-code commands to switch on and off.

It looks like the code that makes these work is in the Grbl Master file.  Since this is an 'exapandable' card, there must be some directions somewhere on how to do it.  SparkFun (seller of the card) pointed me here to GRBL Github.  Am I in the wrong place?

jahnj0584, 2016-07-03:
Can you wire the fan to the spindle pins and use that command while doing
the spindle router manually

chamnit, 2016-07-03:
@subdude27196 : You are in the right place in terms of Grbl. However, Reprap is a name of a 3d printing project, not particularly a brand, and it's not associated with Grbl in any way. 

As far as the RepRap controller card working specifically with Grbl, I have not heard of such a thing or what command they would use to make it work. Again, there is no command Grbl accepts that will explicitly turn a fan on or off. But, it can use other similar supported commands like coolant on/off or spindle on/off. 

I suggest you read through the Grbl wiki (link on front page) to familiarize yourself with what Grbl can do, while keeping in mind that Grbl does not explicitly work with 3d printing hardware or Arduino hardware.

lalo-uy, 2016-07-04:
looking into the Sparkfun schematics and PCB files, the fan connector goes
directly to 12V , no way to switch it on / off