grbl/grbl #1035: Grbl for laser engraver

tyfr12203, 2016-07-02:
I'm trying to make a 2 axis laser engraver. I've read all of the old post on this issue but cant seem to figure out how to get grbl to forget the z axis. I've tried all the recommended code changes compiled and uploaded but it never seems to work. Every  time I home it i hear a high pitch noise and then it says homing failed. Is there a grbl code already written for a laser engraver? Or any other options? thanks

chamnit, 2016-07-03:
@tyfr12203 : Make sure you are editing the config.h file inside the Arduino's library folder, not the folder you downloaded. It's a common mistake.

tyfr12203, 2016-07-03:
I'm editing it in the folder I downloaded and then renaming it and uploading it as a new library then compile and upload. Is that the same thing? Also I'm using universal g code sender when trying to home.

chamnit, 2016-07-03:
@tyfr12203 : Depends on if the Arduino IDE is compiling the new config.h file or not. All you have to do is edit the homing cycle such that
```
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```
or
```
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
```
Compile and flash. You're done. If it's still trying to home the Z, it's not reading the correct file.

closed 2016-07-03 by chamnit