grbl/grbl #1037: Probe not recognized

jahnj0584, 2016-07-11:
My probe doesnt want to work in GRBL 0.9j! I have one pin to A5, and the other to GRD on the same row. One clip on a metal block, other on the end mill. I inverted the pin ($6) to make it actually work, and send g38.2 Z-10 F30 and it just drops the Z slowly. Even touching the wires together wont trigger it, so clearly something is wrong with my inputs. Any ideas?

jaketeater, 2016-07-11:
I don't think that you want to invert the pin, and then try to ground it.  Either raise it to 5v with your current config, or go back to $6=0 and then ground it as you are doing.  

> **By default, the probe pin is held normally-high with the Arduino's internal pull-up resistor. When the probe pin is low, Grbl interprets this as triggered.** For the opposite behavior, just invert the probe pin by typing $6=1. Disable with $6=0. You may need a power cycle to load the change.
> 
> NOTE: If you invert your probe pin, you will need an external pull-down resistor wired in to the probe pin to prevent overloading it with current and frying it.

jahnj0584, 2016-07-11:
well, if I keep it on $6=0 I just get a probe alarm. Setting it to 1 was the only thing that made it actually move.

jaketeater, 2016-07-11:
You probably need a pull up resistor, similar to what is used on limit switches.

jahnj0584, 2016-07-11:
see I dont think I do, because there are many videos and forums of people just using 2 wires on clips, UNLESS they keep hiding that part?

jaketeater, 2016-07-11:
Since it is inverted, try raising the A5 pin by connecting the other wire to the 5v pin instead of ground.

chamnit, 2016-07-11:
@jahnj0584 : With $6=0, a probe error before it start motion occurs when the probe pin is reading a low signal or "triggered". With $6=1, this low signal means the switch is not triggered and good to start the probe cycle, which is why it's working. Check your probe pin wiring. You either are accidentally connecting the pin to ground or some floating state that makes the pin read low. Also, it's advise to place a filter on your input pin to avoid noise issues.

closed 2016-07-11 by chamnit

jahnj0584, 2016-07-11:
ok, but are A5 and GRD the correct pins on the gshield? I know it switched with 0.9 but i have a jumble of numbers in my head from all the forum posts I read before asking here.