grbl/grbl #1039: cycle saving in critical path

J-Dunn, 2016-07-12:
probe.c says: 

```
// Monitors probe pin state and records the system position when detected. Called by the
// stepper ISR per ISR tick.
// NOTE: This function must be extremely efficient as to not bog down the stepper ISR.
void probe_state_monitor()
{
  if (sys_probe_state == PROBE_ACTIVE) {

```
since probing is very rarely needed, this call is pointlessly loading the critical path in the stepper ISR. This test should be done BEFORE calling the procedure not inside it.

suggest stepper.c 
```
  // Check probing state.
   if (sys.probe_state == PROBE_ACTIVE)  probe_state_monitor();
```

I have not done a byte count but avoiding the call overhead would be good in the critical path.

chamnit, 2016-07-12:
True. I think I did it this way so that probe specific tests would be done in the probe.c file, rather than in the stepper.c file. Also, I think this becomes inlined by the compiler anyhow, so it shouldn't effect performance.

J-Dunn, 2016-07-12:
It does not seem to be inlined : 

```
    4350:	39 c1       	rjmp	.+626    	; 0x45c4 <__vector_11+0x450>
    4352:	0e 94 ed 09 	call	0x13da	; 0x13da <probe_state_monitor>
    4356:	10 92 a0 04 	sts	0x04A0, r1
    435a:	10 92 9f 04 	sts	0x049F, r1

```
```

000013da <probe_state_monitor>:
    13da:	80 91 4e 05 	lds	r24, 0x054E
    13de:	81 30       	cpi	r24, 0x01	; 1
    13e0:	a1 f4       	brne	.+40     	; 0x140a <probe_state_monitor+0x30>
    13e2:	0e 94 e7 09 	call	0x13ce	; 0x13ce <probe_get_state>

```

IIRC that will only happen if it is in the same file.  

At least there is no need for pushing and popping registers here. 
The little extra does not matter when probing so a separate file and a call does not matter in that case. 

A while ago you were asking for anywhere anyone saw some speed optimisation in the ISRs , so this is one suggestion.

chamnit, 2016-07-12:
@J-Dunn : Thanks for suggestion and checking the inline. I'll put this improvement into v1.0 and leave v0.9 master as is.

closed 2016-07-12 by chamnit

splitn2, 2016-07-13:
I'm starting a chant , 1dot0 1dot0 1dot0 1DOT0 1DOT0 1DOT0 ONEDOTZERO ONEDOTZERO ONEDOTZERO **ONEDOTZERO ONEDOTZERO ONEDOTZERO**, its getting louder and louder!!!!! Keep up the great work @chamnit I really appreciate your effort, I'd love to be an early adopter if there is a way? cheers, Richard

J-Dunn, 2016-07-22:
Another 6 byte saving in function called from within  ISR(SERIAL_RX), makes it quite a bit easier to read too. 

```
uint8_t serial_get_rx_buffer_count()
{
/*  uint8_t rtail = serial_rx_buffer_tail; // Copy to limit multiple calls to volatile
  if (serial_rx_buffer_head >= rtail) { return(serial_rx_buffer_head-rtail); }
  return (RX_BUFFER_SIZE - (rtail-serial_rx_buffer_head));
*/
//  uint8_t rtail = serial_rx_buffer_tail; // Copy to limit multiple calls to volatile
  uint8_t diff = serial_rx_buffer_head -serial_rx_buffer_tail;
  if (diff >= 0) { return(diff); }
  return (RX_BUFFER_SIZE + diff);
}

```
Same thing can be done in  serial_get_tx_buffer_count()  though that is usually unused and optimised out.