grbl/grbl #1040: all limit switches working on all axis

466, 2016-07-18:
Hi there,
i just made some (4 actually, x & y) limit switches for my shapeoko 2 an wired them up (https://lh4.googleusercontent.com/-FhXABwKcb-I/UvMhBLTESlI/AAAAAAAADCI/kfE-f7jkrwc/s800/cnc_limit_switch_c3.png) , now i do have some questions left.

I did activate the switches via:
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)

When i move in x direction, and press the X-limit switch, grbl enters alarm mode and stops, just as expected.
When i move in x direction and press the Y-limit switch, the very same happens, is this the correct behavior ^^?
If so, does grbl it self recognize this as different limit switches? (As there are x-, y- and z-limit switch pins)

Looking forward to some help, best regards
466

langwadt, 2016-07-18:
sounds like normal operation, differentiating between switches  is only done (and needed) when homing
during normal operation all limit switches mean the same, something bad happened stop.

466, 2016-07-18:
Greate, thank you!
I was not sure about this, glad to hear everything works as expected!

Looking forward using these now; will save a LOT of time!

466, 2016-07-23:
I got homing working, no problems, at least for  2 days ... 

I did change some wires within my setup (hiher quality) , measured all limit switches, working just like yesterday, nothing 'really' changed.
But grbl is not able to do a homing cycle anymore :(

I start homing and it does the z-axis homing: done
After that x- and y- axis are next.
Now, the problem is, grbl does stop, once it hits ONE switch, where it does not matter weather this is x- or y- switch, it stops and pulls of  minus 1mm from x- and y- home.

This does look like x- and y- switch being triggered at the same time; but as this is not the case... .
Any ideas?

jahnj0584, 2016-07-23:
double check your wiring, make sure all the pins are in the proper layout.

electrokean, 2016-07-24:
Have you got any filtering on your limit switches?
A 0.1uF (100nF) capacitor between each limit signal and ground will often remove false triggering due to noise.

closed 2016-09-03 by chamnit