grbl/grbl #1041: Motors only drives in one direction

rippschemitkraut, 2016-07-19:
i have a really strange problem with grbl.
im using a arduino mega board and connected everything like usual but can only drives all axis in one direction. i checked the dir pins on the mega board and there is some 0.900v when the axis dont move and when i move the axis it changes to 1.3v. all axis are different. x begins by 0.4v without moving and with moving 0.8~0.9v. y begins with 1v. while moving to 1.4v and so on. inverting dir pins does not the trick nor inverting pulse pin.imotors just only drives other direction but only in one direction.
 motor cabling cant be wrong because it runs before on db25-1205 bob under linuxcnc. even disabling endstops wont help. is there anything what i have missed ?

here is my config
$0=50 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.010 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=2133 .000(x, step/mm)
$101=2133.000 (y, step/mm)
$102=2133.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=1000.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
the arduino mega r3 is a china clone with ch340g. 
drivers are m542
motors are  23HS9430

maybe someone has the same problem like me. 

regards

jahnj0584, 2016-07-19:
You have some weird values for steps and acceleration. Is that intentional?

rippschemitkraut, 2016-07-20:
i calculated it acording to the 1/32 microsteps and my 12x3 leadscrew using http://prusaprinters.org/calculator/

is that wrong ?

rippschemitkraut, 2016-07-20:
i have measured the voltag on the pulse +. without moving it shows 0v and with moving it show 2v. on dir + what i said before that without moving it show 1v and while moving it shows some 0.4v more but it must be 5v i think or im wrong ?

rippschemitkraut, 2016-07-20:
new day, new try but no luck.
have measured the pulse + and when its not moving it shows me 0v and while moving it shows 2v. is that ok or correct ? swapping this cables with the dir doesnt do anything.
Connection on the mega looks like this picture
[image]
all other wire like pulse - , dir - and enable - goes to gnd.

is there something what i have missed ? regards

rippschemitkraut, 2016-07-20:
have found the error. can be closed. it is the 1. picture what is wrong. so it was my mistake.  for not reading properly the post where i found this images.

Btw. im using for controlling a self build brushless spindle a small maxon 1quadrant controller DECS 50/5 with 48v power supply. it does not work with pwm but digital 10 bit with a resolution of 1024 and 0~5v. 
how i can add this to grbl ? any advice ? thank you

closed 2016-09-03 by chamnit