grbl/grbl #1043: homing direction and resulting machine position

christopherwit, 2016-07-20:
Dear recently I'm trying to figure out axis direction, homing direction and the resulting machine position and soft limits.

I tend to be missing one parameter to set (this can be fixed through grbl firmware editing but I would like to hear if others agree.

My setup is the following:
 - three end stops at x0 y0 and z0. 
 - homing pull of, of 4 mm
 - max travels set to x177, y200 and z24.

What I like to be able to set is homing in the negative direction, moving to x0,y0,z0 and pull of to x4, y4 and z4. 
I keep getting negative values of -4 or max travels-4 like x-173, y-196 and z-20. Or the homing direction is the wrong way around, or the axis direction are inverted.

To me this sounds like even if the homing direction is inverted by a mask the firmware still assumes traveling to the extremities of the allowed travel distances, but never back to 0,0,0.

Could someone help me out on getting the right configuration, or support me in getting an additional setting to make the homing travel to 0,0,0 ?

Current settings:
$0=10 (step pulse, usec)                                                      
$1=25 (step idle delay, msec)                                                 
$2=0 (step port invert mask:00000000)                                         
$3=4 (dir port invert mask:00000100)                                          
$4=0 (step enable invert, bool)                                               
$5=0 (limit pins invert, bool)                                                
$6=0 (probe pin invert, bool)                                                 
$10=19 (status report mask:00010011)                                          
$11=0.010 (junction deviation, mm)                                            
$12=0.002 (arc tolerance, mm)                                                 
$13=0 (report inches, bool)
$20=0 (soft limits, bool)                                                     
$21=1 (hard limits, bool)                                                     
$22=1 (homing cycle, bool)                                                    
$23=0 (homing dir invert mask:00000000)                                       
$24=25.000 (homing feed, mm/min)                                              
$25=500.000 (homing seek, mm/min)                                             
$26=250 (homing debounce, msec)                                               
$27=4.000 (homing pull-off, mm)                                               
$100=56.000 (x, step/mm)                                                      
$101=51.000 (y, step/mm)                                                      
$102=50.000 (z, step/mm)                                                      
$110=500.000 (x max rate, mm/min)                                             
$111=500.000 (y max rate, mm/min)                                             
$112=500.000 (z max rate, mm/min)                                             
$120=10.000 (x accel, mm/sec^2)                                               
$121=10.000 (y accel, mm/sec^2)                                               
$122=10.000 (z accel, mm/sec^2)                                               
$130=177.000 (x max travel, mm)                                               
$131=200.000 (y max travel, mm)                                               
$132=24.000 (z max travel, mm)
? result for homing: <Idle,MPos:-4.000,-4.000,-4.000,WPos:-4.000,-4.000,-4.000,Lim:000>

langwadt, 2016-07-20:
when you say your switches are at 0,0,0 do you then mean lower left corner, Z down or 
upper right corner, Z up ?

christopherwit, 2016-07-21:
Since I wanted a right turning axis system, and starting with z=0 on the top of my object, resulting  in everything I cut away has a positive increasing z coordinate:
My axis are in the left rear corner, with y pointing towards yourself, with x to the right and z pointing down.
Switches are at the rear, left and top.

chamnit, 2016-07-25:
@christopherwit : Please read this [FAQ answer](https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to).

christopherwit, 2016-07-26:
Thanks Sonny for pointing me out this faq item. I was aware of the
differences between printing and milling. When I get back from holiday I
will have a proper look at it.

<https://github.com/grbl/grbl/issues/1043#issuecomment-235007146>, or mute
the thread

closed 2016-09-03 by chamnit