grbl/grbl #1044: Using 0.9i G02 Question

rfresh737, 2016-07-22:
Can use G02 to make a full circle? I read somewhere (now I can't find it) that I should make circles by breaking them down into 4 parts and make them with 4 arcs.

I'm getting some wood surface burn marks at the 12, 3, 6 and 9 o'clock position of my circles because I am creating using 4 arcs.

Using 0.9i is it safe for me to create a circle using one G02 command?

Thanks...

109JB, 2016-07-25:
In general, the problem you are referring to is for radius format arcs which aren't supported in grbl. Center format arcs should not suffer from this problem and I have not noticed it in grbl. Having said that, Sonny should be able to tell if there are any peculiarities in grbl arc implementation that I am not aware if.

chamnit, 2016-07-25:
@rfresh737 : Correction. Grbl does support arcs in R radius mode. It's just not recommended to use it. The radius-mode arc description itself is very poorly formulated, such that numerical errors in certain situations are awful.

To draw full circles, I think all you need to do is specify the target point as the current location and supply the IJK offset. Or, just the IJK offset. I can't remember at the moment. 

Anyhow, it shouldn't matter anyway. If you are streaming the g-code, Grbl should move through 4 quarter-circle arcs smoothly without any pausing. If you are hand entering the circles, you will have that pausing problem.

closed 2016-07-25 by chamnit

rfresh737, 2016-07-25:
Thanks for the clarification. I'm not using the Radius method. It looks like circles are working fine with being drawn in one G02/G03 line so I am using that method (I dropped making full circles using 4 arcs).
The pausing issue has gone away now.