grbl/grbl #1046: coreXY PWM Laser

tom4tom, 2016-07-27:
@chamnit: thank you for fixing coreXY homing an compiling. It works like a charm!
Now I have the Problem, when I use PWM for controlling the Laser (S commands in every Pixel/G-Code Line) , the coreXY mouve nor fluent..... 
Any Idea?

chamnit, 2016-07-27:
@tom4tom : That will be a feature of v1.0. Hold tight.

closed 2016-07-27 by chamnit

splitn2, 2016-07-27:
1.0 1.0 1,0 !!!!! keep up the good work Sonny thanks

tom4tom, 2016-07-27:
Ah, cool! Tanks Sonny.

tom4tom, 2016-08-02:
Hi Sonny, can you make a statement, when we can expect this feature? (Only to have an orientation)

chamnit, 2016-08-02:
@tom4tom : Soon. There are higher priority things outside my control that have been consuming my time over the past year, such as work but mainly family health issues. Fortunately, they look like they are starting to subside. So please be patient. It'll be out when its out.

tom4tom, 2016-08-02:
Hi Sonny, thank you for your Statement.
Health is the most important thing in Live. Take care of your Family, and take the time that Things need....

splitn2, 2016-08-02:
agreed family and health is number one,  hope all is well Sonny. cheers

tom4tom, 2016-09-19:
Hi Sonny, any news here (1.0 coreXY)?

chamnit, 2016-09-19:
@tom4tom : Major GUI developers are receiving a pre-release this week to update their code to make it compatible. The delay has been due to a change in the status reports. It had to be modified to contain more information and still make it smaller.

tklus, 2016-09-19:
So excited for 1.0 :D

splitn2, 2016-09-19:
Awesome!!!!!

tom4tom, 2016-09-20:
Hi Sonny, do you have this GUI People on the list, for the pre-release?
- https://github.com/Denvi/grblControl
- I2LEG great for Laser, no Website, no GIT: thesava@gmail.com
- http://www.picengrave.com/Pic%20Programs.htm 
- https://jtechphotonics.com/?page_id=1980

jorgerobles, 2016-11-09:
Hi, I've got a GRBL 0.9j with CoreXY enabled running a laser and get the same movement issues during raster engraving.
I know is fixed in 1.1d. but my build is running on a Nano board, and cannot compile 1.1d on that thing. Is there any chance to get a backport on 0.9 branch just for the laser movement patch?

Thanks in advance!

chamnit, 2016-11-09:
@jorgerobles : No. v1.1 laser mode is tightly integrated with spindle speed overrides, because they generally work the same. I don't intend to make and support separate versions.

v1.1d is right on the edge of fitting on an FTDI-based Arduino, like your Nano. With CoreXY enabled, you will likely exceed that limit. One alternative is to write the Arduino Uno boot loader on your Arduino Nano. It's not as hard as it may sound. If you have a spare Arduino, you can use it to write the boot loader.

- Wire your spare Arduino to your Nano as an ISP. (https://www.arduino.cc/en/Tutorial/ArduinoISP)
- Connect your spare Arduino to your computer and fire up the Arduino IDE.
- Select 'ARDUINO UNO' as the board. This is very important. 
- Select 'Burn Bootloader'  in the Tools menu.

Now your Arduino Nano will have an extra 1KB of flash space to fit your CoreXY code. Just make sure you select 'Arduino Uno' as the board whenever you flash your modified Nano.

I've done this operation to every Nano I own and use them as portable development boards. It works., but you may have to physically press the reset button right when the flashing starts. It can sometimes timeout and not start. Manually reseting helps kick start it. I've found this is the case on only one of my Nanos.

jorgerobles, 2016-11-09:
Thanks, indeed I've used optiboot, freeing 1.5kb, but once is written I got
the Nano stuck. I will try with the uno bootloader anyways.

Thanks!