grbl/grbl #1047: streamlining interrupts

J-Dunn, 2016-07-29:
The current timer setup works fine and is well tested, but I'm wondering whether it could be simplified and thus speeded up. 

Are there supported driver hardware contexts which require a fixed duration STEP pulse, or could this be reduced to a more flexible condition of minimum time for STEP to be asserted? 

Thanks.

langwadt, 2016-07-29:
I don't see why any hardware would require anything but a minium time, but as the time spend in the interrupt approaches the step period you run out of "off" time. 
And the same the interrupts provide the option to add delay from setting the directions to the step pulses some hardware might need that

The directions could be set at the end of the stepper interrupt, but then it is back to running out of "off" time

gerritv, 2016-07-29:
Is there a particular reason to change what is working? There are risks involved in any change.

J-Dunn, 2016-07-29:
"Is there a particular reason to change what is working? "

This came up as part of porting to STM32 where the different interrupt structure means it will get changed anyway. I posted the question here as the place where relevant knowledge of the hundreds of different machine hardware configs is. 

My guess was the same as what  langwadt said but that may just be lack of knowledge of some of the possible permutations and requirements.

usbcnc, 2016-07-29:
@J-Dunn ,If you want to experiment the timer, you can try the Windows simulation project. This will help you a lot to understand how it works. I am using it now to analyze the timer and see if it is possible to optimize the timer and even put s-curve to GRBL. Windows simulation can help you run and debug the code without hardware. 
What I am trying to first to see if it is possible to remove the srqt function for each timer calculated. Latter I might want to think using integer only to calculate the circle. No need to use float point and sin cos functions which not only save time but also save a lot of code space.

J-Dunn, 2016-07-30:
`What I am trying to do first to see if it is possible to remove the srqt function for each timer calculated. `
It is hard to see how you can avoid the need for sqrt in this kind of calculation. I think it may be worth looking at using a Taylor series to the required precision but I doubt whether it is needed on 32b hardware with an FPU.

langwadt, 2016-07-30:
I roughly timed a sqrt(float) on a 100MHz STM32M4, it takes ~10us, the math lib doesn't use the FPU instruction probably to be strictly IEEE compliant? 

the asm instruction vsqrt.f32 takes 14 cycles so 140ns

redefining sqrt() as a function the uses the  vsqrt.f32 doesn't seem to break anything in a quick test

J-Dunn, 2016-07-31:
Wow, two orders of magnitude faster. I'd assumed that FPU would be being used if it has a sqrt instruction. Probably best to define it as an inline macro rather than adding the overhead of a fn call.

It's mostly used in the planner stage which is less critical but seems worth using if it's there. 

If it is taking 10us on STM32 , it must be at least 60us on AVR. 

limits.c:  
 `   st_prep_buffer(); // Check and prep segment buffer. NOTE: Should take no longer than 200us.
`
It looks like sqrt is significant part of that 200us. This is probably worth replacing with a four term Taylor series expansion on AVR. That would probably reduce it to a few us.

J-Dunn, 2016-07-31:
> math lib doesn't use the FPU instruction probably to be strictly IEEE compliant? 

I found an errata paper from ST saying there is an issue if VSQRT is used in an ISR which returns before the 14 cycles is complete. This may have been a reason not to use it in IEEE context.

langwadt, 2016-07-31:
think I found the reason, AVR-GCC has the quirk that floats and doubles are the same so it doesn't matter that sqrt(double) is used instead of sqrtf(float) they call the same function

on ARM float is 32bit, double is 64 bit so when using sqrt() the lengthy lib function that can handle 64bit is used, sqrtf() compiles to a single vsqrt.f32 instruction with anything but -O0

for porting the code should probably be changed to explicitly use the float version of the floating point functions, i.e. sinf,cosf,fabsf etc.

J-Dunn, 2016-07-31:
Thanks langwadt,  just for completeness CMSIS does not seem to use sqrtf() but has it's own fastmath library with arm_sqrt_q15() et. cie.

langwadt, 2016-07-31:
much simpler,  in grbl.h replace math.h with tgmath.h it is aware of the types and call the right functions

J-Dunn, 2016-07-31:
Thanks, I was not aware of that header. Buillding with that knocked about 2k off the size of the elf. 

However, it does not seem to be getting me a VSQRT :

080170d4 <sqrtf>:
 80170d4: b510        push  {r4, lr}
 80170d6: ed2d 8b02   vpush {d8}
 80170da: 4c29        ldr r4, [pc, #164]  ; (8017180 <sqrtf+0xac>)
 80170dc: b08a        sub sp, #40 ; 0x28
 80170de: eef0 8a40   vmov.f32  s17, s0
 80170e2: f000 fb23   bl  801772c <__ieee754_sqrtf>

langwadt, 2016-07-31:
try turning on some optimization, like -OG

chamnit, 2016-07-31:
@J-Dunn: FYI, float multiplies on the AVR are not so fast. They take 9-10us for each float multiply (float divisions are way worse). You may get some more cycles by using a taylor expansion, but I'm wary of the error induced by its approximation and how I may make dependent calculations unstable. You have to tread very carefully when you do things like that. That said, I don't think it's worth the time and effort to glean a few more cycles here and there for the 328p, since it's operating ok for most people as is. As the old saying goes, if it isn't broke, don't "fix" it.

gerritv, 2016-07-31:
If you really feel strongly to spend time optimizing things, then you should be using something like: https://mcuoneclipse.com/2015/08/23/tutorial-using-gnu-profiling-gprof-with-arm-cortex-m/. Anything else is stabbing in the dark. I have gone up this path many times in the past, almost always to find that where I thought there was a problem isn't where it actually is. In the end I would have been better off spending my energy on adding functionality :-( It is however a good educational experience to dig as deep as you are.

J-Dunn, 2016-07-31:
As I pointed out in #4 above, the main point here is in porting to STM32 and the question was does any supported hardware need a fixed pulse period rather than a simply a minimum period with STEP asserted.

J-Dunn, 2016-07-31:
```
         prep.exit_speed = sqrt(pl_block->entry_speed_sqr-2*pl_block->acceleration*pl_block->millimeters);
 8011fc4: edd3 7a09   vldr  s15, [r3, #36]  ; 0x24
 8011fc8: ee77 7aa7   vadd.f32  s15, s15, s15
 8011fcc: ee67 7a87   vmul.f32  s15, s15, s14
 8011fd0: ee76 7ae7   vsub.f32  s15, s13, s15
 8011fd4: eeb1 0ae7   vsqrt.f32 s0, s15
 8011fd8: eeb4 0a40   vcmp.f32  s0, s0
 8011fdc: eef1 fa10   vmrs  APSR_nzcv, fpscr
 8011fe0: d009        beq.n 8011ff6 <st_prep_buffer+0x186>
 8011fe2: edd3 7a09   vldr  s15, [r3, #36]  ; 0x24
 8011fe6: ee77 7aa7   vadd.f32  s15, s15, s15
 8011fea: ee27 0a87   vmul.f32  s0, s15, s14
 8011fee: ee36 0ac0   vsub.f32  s0, s13, s0
 8011ff2: f001 fd2b   bl  8013a4c <sqrtf>

```

Ah, I think I found why sqrtf is still in there. Since VSQRT returns zero on error as well as a non erroneous zero result, the compiler has to fall back to sqrtf to get error handling. So it's usually the very fast FPU call but the sqrtf and  __ieee754_sqrtf code is still compiled in.

closed 2016-09-03 by chamnit