grbl/grbl #1048: Should $C  lose state?

cspwcspw, 2016-07-31:
The $C toggle on (Check-mode-on) preserves the parser state.  (You can check parser state with $G).  But the $C (Check-mode-off) does a soft reset, and loses parser state.  So if we have set a working coordinate system (say G55), after the sequence $C (Check-mode-on) and then another $C (Check-mode-off), we've lost our G55 setting.

Is the reset on Check-mode-off by design, and if so, what is the rationale?  I think I'd prefer to turn off Check mode without a soft reset, but I'm not sure I fully understand the implications.

Thanks

chamnit, 2016-07-31:
@cspwcspw : Yes. Disabling check mode will reset the modal state. This always brings Grbl back to a default state, where the modes and the offsets are always known and never dependent on what the check mode did. In manufacturing, this is a requirement, if not always, on machines and helps prevent operator error and **very** expensive crashes.

closed 2016-07-31 by chamnit

cspwcspw, 2016-07-31:
Okay, I can see that.   But then should entering check mode not also reset the modal state?

It seems inconsistent that I can check my code and it might succeed because of residual modal state, then I run it and millimeters become inches, or I've forgotten to set the work coordinates, so I still get the very expensive crashes.

chamnit, 2016-07-31:
@cspwcspw : It's advised to always start your job with a reset. So that it's consistent. Your CAM program should be setting your initial modal states in the g-code header. This first line is typical of CAM programs. It's a bunch of g-code states that are there to safe guard against problems like forgetting you set one earlier. If your CAM program doesn't do this, add one. 

The only things that are lost during a reset, except for the modal states, are the G92 coordinate and G43.1 tool length offsets. For this reason, it's advised to not use G92 at all but use G54-59 work coordinates. G43.1 should only be used if your GUI tracks the tool length offsets for you or you hard code them in.