grbl/grbl #1049: Brushless spindle and PWM

fantasiiio, 2016-08-05:
Hi,
I'm using grbl and I like it ! 

Currently, I'm using a RC brushless motor with ESC and a servo tester (PWM generator) to set the motor speed.

I was wondering if the PWM output of GRBL is compatible with any RC ESC, with a frequency of 50hz ?
Or, is there something I can do in the configuration and/or code ?

chamnit, 2016-08-05:
@fantasiiio : Grbl outputs a 7.8 kHz PWM signal (if I remember right) of varying duty cycles depending on spindle speed. Or rather a voltage level. If you need something else, you'll need to alter pin output source code in spindle_control.c or make an adapter that converts Grbl's signal to your desired signal. The latter is probably easier.

closed 2016-08-05 by chamnit

fantasiiio, 2016-08-05:
I played with pwm, with the datasheet and examples and the conclusion is, we can only use the pin 3  (Output B of timer2) if we want to modify the frequency of the pwm...  so this is not possible with hardware pwm with current pins configuration.

I was alsost there.. I have a 50.1hz pwm, but on pin 11, the duty cycle is stuck to 50% in this mode.. damn it.

http://www.righto.com/2009/07/secrets-of-arduino-pwm.html
section: Varying the timer top limit

langwadt, 2016-08-05:
I don't use an AVR but wouldn't setting the prescaler to 1024, MINIMUM_SPINDLE_PWM roughly 15, PWM_MAX_VALUE roughly 23, give you something rather coarse but usable?

~1ms  to ~1.5ms, 61Hz update rate

fantasiiio, 2016-08-05:
I know... I didnt tried it because i tought it would be too much. I will give it a try later.

langwadt, 2016-08-05:
I think most ESCs will handle several hundred Hz with no problem

fantasiiio, 2016-08-06:
I tried it. It's kind of working... I have a 8 speed motor.. Lol

Also, the minimum pwm simply chop the duty if its less than minimum. So i have 60% to 100% of throttle. Less than that is always to 60%.

I'd prefer to make a converter module with another arduino that use analogread and servo write..

fantasiiio, 2016-08-06:
I made a signal converter for my esc and it works. I can now control my rc brushless motor with grbl.

Thanks for your support !

Macioza, 2016-08-07:
@fantasiiio : How does your converter look like? I'm also trying to develop a Arduino-based middleman for a RC brushless spindle with a ESC. Manual control via potentiometer works fine, but gcode controll would be great. Measuring a PWM wave with a Arduino seems impossible, so what are the options?

langwadt, 2016-08-07:
you can use the input capture to measure duty cycle, but why bother making it complicated
every ~20ms poll the input in a loop to get the high period, then set an output high for the corresponding 0.5-1.5ms

electrokean, 2016-08-07:
@langwadt even easier, you can use a low pass filter to convert the pwm output to a voltage that the Arduino can read via analogRead()
Google "pwm low pass filter" for info - top result looks good:
http://www.allaboutcircuits.com/technical-articles/low-pass-filter-a-pwm-signal-into-an-analog-voltage/

langwadt, 2016-08-07:
I'd say a few lines of code are simpler than and external filter, unless you can take a dirt cheap servo tester and replace the potmeter with an analog input

fantasiiio, 2016-08-08:
@Macioza : Yes, I made a converter that read pwm signal and generate another one. There is one problem left with it. It does detect 0% (OFF) and 100%. I have to put some code to check this. But I can share the code tomorrow. I'm also coding a RPM meter in the same code.

swarfer, 2016-08-08:
This was done a while back...
http://openbuilds.com/builds/software-speed-control-of-a-brushless-dc-bldc-motor-from-mach3.762/
I don't know if that one deals with the flat line 0% and 100% states that GRBL outputs.

fantasiiio, 2016-08-08:
@swarfer : Cool, I did not see that before. But I'd prefer using a simple arduino mini without external components. I don't have any pic mcu.

But the idea of manual pwm option is very good.
Thanks for the link.

BlueOrangeLive, 2016-08-08:
Hello
I had tested recently for me as an ESC brushless spindle installed
in grbl V1. and gnea / grbl-Mega.
https://github.com/BlueOrangeLive/GRBL_328_BrSpindel
Spindle parameters can be adjusted in default.h.
Greeting Jürgen

Hallo
Ich hatte vor kurzem für mich test weise eine ESC-Brushless Spindel eingebaut 
in grbl V1. und gnea/grbl-Mega . 
https://github.com/BlueOrangeLive/GRBL_328_BrSpindel
Spindel Parameter können in default.h angepasst werden.
Gruß Jürgen

fantasiiio, 2016-08-08:
@BlueOrangeLive : I saw it, and it disapeared...

BlueOrangeLive, 2016-08-08:
@fantasiiio 
its now back again.

mschorer, 2016-08-08:
Also there is
https://github.com/mschorer/grbl_mc_leonardo

BlueOrangeLive, 2016-08-08:
My ESC version is integrated into GRBL. I added a couple of lines of code. Thus it can be an ESC-controler dirkt it to drive.
When spindle speed above zero then rotates at a minimum speed. From a higher speed than the minimum speed it is then faster.
I use bCNC with Linux as a control program.
Here is a short video with my GRBL Nano Shield without connected step motore.
https://www.youtube.com/watch?v=LzEDxabJi_Q

https://github.com/BlueOrangeLive/GRBL_328_BrSpindel

fantasiiio, 2016-08-08:
I didn't know a esc will works at 488hz... you have a little more resolution than my test at 60hz. You have something like 128 to 253 for pwm value (1ms to 2ms), instead of 15 to 23...

I'll give it a try.

motanescu, 2019-01-16:
Get a second arduino and load the servo sketch on it. Also get the CCPM tester. Then feed any 5V PWM into A0 and output 50HZ 1-2ms from Pin 9. Costs $2 and takes no more than 5 minutes. Messing with grbl settings just isn't worth the hassle imo

urpylka, 2021-09-27:
I made some fixes of [kirilnedev/grbl-rc-pwm](https://github.com/kirilnedev/grbl-rc-pwm) in my repo [urpylka/grbl-rc-pwm](https://github.com/urpylka/grbl-rc-pwm).

Also about it I wrote the article about options how to connect ESC to `grbl-controller` by the Arduino converter [urpylka.com/posts/post-36/](https://urpylka.com/posts/post-36/) (in Russian).

gururajkashikar, 2023-02-11:
Hi All, Urpylka

As many of you are suggesting, I do not want to disturb my working Grbl 1.1 and CNC to run BLDC spindle. Hence I used yet another Arduino Nano with code from Urpylka's  grbl-rc-pwm git repo.

Urpylka, I need your advise on setting few things.

If I use the Arduino code from [here ](https://howtomechatronics.com/tutorials/arduino/arduino-brushless-motor-control-tutorial-esc-bldc/) with a POT, Spindle is spinning perfectly.

But when I upload your code from grbl-rc-pwm git repo, BLDC A2212/6T, 2200KV just makes ticking noise but not Spinning. 

Following are my settings...

1.  PWM pin from CNCShield connected to Pin 2 of BLDC Arduino. Note : This pin is working with my 500W Chinese Spindle.
2. ESC's PWM input is connected D4 as per spindlePwmPin, ESC Gnd-> Arduino Gnd, ECS power pin (Red) -> Vin of Arduino.
4.  Have tired to set $30 == 7000 and $31 == 3890. Even tried $30 == 24000 and $31 == 0
5.  Then issued M03 S250, S1000, S4000, S5000, S7000.... But Spindle does not spin. 

Please suggest if I am missing something here to successfully control the speed of BLDC motor from grbl1.1's PWM

Thanks in advance.

urpylka, 2023-02-14:
@gururajkashikar Hi, I don't have exact answer. I suggest you to debug step by step. It will make clear the situation.

1. Check the output pin (spindlePwmPin) of `grbl-rc-pwm` by oscilloscope
2. If it exist but don't work. Then compare it with [worked solution](https://howtomechatronics.com/tutorials/arduino/arduino-brushless-motor-control-tutorial-esc-bldc/)
3. If it doesn't exist (or doesn't work correct you can add debug inforamation to the code to blink LED or write smth to UART
4. Make calibration of ESC (if needed)

HenriqueChieza, 2024-12-21:
Olá pessoal, tudo bem? Eu consegui resolver esse problema, mais ou menos, eu abri o arquivo "cpu_map.h" e adicionei essa parte: // Prescaled, 8-bit Fast PWM mode.
    #define SPINDLE_TCCRA_INIT_MASK   ((1<<WGM20) | (1<<WGM21))  // Configures fast PWM mode.
    // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20)               // Disable prescaler -> 62.5kHz
    // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS21)               // 1/8 prescaler -> 7.8kHz (Used in v0.9)
    // #define SPINDLE_TCCRB_INIT_MASK   ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
    // #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98kHz (J-tech laser)
    **#define SPINDLE_TCCRB_INIT_MASK      ((1<<CS22) | (1<<CS21)) // 1/128 prescaler** -> 0.49kHz (BLDC ESC compatible) //------------------AQUI MODIFICADO----------//

ATENÇÃO: ESSE BLOCO SE REPETE 2 VEZES, MODIFIQUE OS DOIS POR PRECAUÇÃO

Hello everyone, how are you? I managed to solve this problem, somehow, I opened the "cpu_map.h" file and added this part: // Prescaled, 8-bit Fast PWM mode.
    #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21)) // Configures fast PWM mode.
    // #define SPINDLE_TCCRB_INIT_MASK (1<<CS20) // Disable prescaler -> 62.5kHz
    // #define SPINDLE_TCCRB_INIT_MASK (1<<CS21) // 1/8 prescaler -> 7.8kHz (Used in v0.9)
    // #define SPINDLE_TCCRB_INIT_MASK ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
    // #define SPINDLE_TCCRB_INIT_MASK (1<<CS22) // 1/64 prescaler -> 0.98kHz (J-tech laser)
    **#define SPINDLE_TCCRB_INIT_MASK ((1<<CS22) | (1<<CS21)) // 1/128 prescaler** -> 0.49kHz (BLDC ESC compatible) //----------- -------HERE MODIFIED----------//

ATTENTION: THIS BLOCK IS REPEATED TWICE, MODIFY BOTH AS A PRECAUTION