grbl/grbl #1050: First Time G4 Dwell Runs, time is not same as second time

baden0001, 2016-08-07:
Running the following code:
G21 G90 G64
G54
G0X23.74Y165.99
G0 Z1.0
G38.2Z-300F1000
G92Z-17
G90G0Z1
M3S10000
G4P1.2
G1 Z1.0 F1000.0
G1 F1200.0 X23.59 Y165.87
. . .(movement removed for issue post)
G1 X23.28 Y163.2
M5
G0 Z10.0
G0 X61.78 Y118.98
G1 F1000.0 Z1.0
G38.2Z-300F1000
G92Z-17
G90G0Z1
M3S10000
G4P1.2
G1 Z1.0 F1000.0
G1 F1200.0 X59.58 Y123.57
. . .(movement removed for issue post)
G1 X30.07 Y105.23
M5
G0 Z10.0
The first G4 in the program does not appear to dwell the correct amount of time.   After this, G4 appears to function consistently.  The G64 at the top of the program doesn't appear to be supported by GRBL, and will be removed for future use.  Wouldn't think that would cause the dwell issue though.

chamnit, 2016-08-08:
@baden0001 : I can't see why there would be any difference in dwell time between all of your examples. A quick test showed that it works fine. Exactly how much of a time difference are you observing?

PicEngraver, 2016-08-08:
Sonny,
I have been seeing the same thing in my coding.  When I use a single G4 P.01 to detect an axis move completion, the code will not halt to wait for move completion.  Had me stumped until @baden0001's post above.   I have just tried adding a second G4 command sequence immediately after the first, and my code routines will wait for the axis move to cease.  I have only just tested briefly today with 9j. 
John Champlain

chamnit, 2016-08-08:
@PicEngraver : I'm having trouble reproducing the issue. I'll need an example to be able to diagnose the problem, if there is indeed one. There's a chance that the buffer synchronize function may not be operating correctly in certain conditions.

PicEngraver, 2016-08-08:
Thanks.  Back soon.
John

baden0001, 2016-08-08:
Planning on documenting more of the issue tonight.  It is on a plasma table and I am wondering if it is in part due to long torch fire times.

PicEngraver, 2016-08-08:
Sonny,
Load the following code: 
(I used both my own sender and UGS to test this.)

G20 G91
F50 G1 X1
G10 P1 L20 X0
G4 P1
F50 G1 X1

Then:
1 - Send the code - should get two distinct 1inch X moves (I'm using 9j)
2 - Click on Reset Zero in UGS
3 - Click on Return to Zero in UGS
4 - Send the code again - almost all the time I get only one X move.

It seems to not be doing this 100%, but often enough to easily have it happen.  I am out of time just now to continue testing, and I have not tried it with two successive G4 P1 commands to see if it then makes the two X moves 100% of the time.   For my present code needs, using two successive G4 P.01 commands eliminates my issues, so that is what I will do for now.
Thank you,
John

baden0001, 2016-08-09:
After looking into the plasma cutter side it does appear that the torch does not fire immediately after grbl sends the on command.  Causing what looks like incorrect dwell.  Consider my side of the issue closed.
I will work around this for now.  
Thanks to the developers that put time into this. Much appreciated!

closed 2016-09-03 by chamnit

chamnit, 2016-09-03:
@PicEngraver : I think your issue is related to the G10 command. When you send this, it has to write some values to EEPROM.  The AVR processor shuts down the interrupts during this time (steppers and serial RXTX included), so it can cause some weirdness. If you move the dwell to in front of the G10 command, your test program runs ok. What I need to do is enforce a stop motion when anything has to write to EEPROM. It doesn't solve everything completely but it does help.

PicEngraver, 2016-09-03:
Thank you Sonny.  I appreciate your time looking at this.
John