grbl/grbl #1051: Homing problem

emteeoh, 2016-08-08:
I'm working with a laser cutter that is built a little oddly: the limit switches are not at edges of travel! The switches are set in 75mm from their edges, in the top right corner.

I'm just starting to switch from the old controller to GRBL, so we want to be able to move the cables back to the old controller if we have any serious problems, so I can't just move the limit switches to somewhere sane.

I've got all the movement working, but the natural position to put the tool-head when loading/unloading is in the lower-left corner (ie: -75,-535), so I set $N0=G28, and then jogged to the lower-left corner and did a G28.1 . My gcode starts with G28\nG92 and ends with a G28. This all seems to work.

The problem I'm having is that the corner where the tool is sitting has the X limit switch set, so if I do a home from there, GRBL gets upset and homing fails. If I do a G92X100 before homing, it works fine.
Is there a way to do the G92X100 automatically, or get GRBL to move off the limit switch *BEFORE* homing?

jahnj0584, 2016-08-09:
If you're already homed why do you home again? You should have a denounce
value after homing anyways so the switch is never on after the initial two
clicks.

emteeoh, 2016-08-09:
I think you misunderstand me. Suppose I'm using the machine today, and everything is fine. At the end of the day, when I'm done, the tool is in the lower-left corner. Yay. I turn everything off in the shop, 'cause that's what we do, and go home.

The next day, I come into the shop and power things up. GRBL resets its MPos to 0,0 , even though its actually at -75,-535, which is fine with me, 'cause I'm about to home it. But then homing fails because when X is in the range of -75 to 0, the limit switch is closed, and GRBL doesn't know how to deal with that. I can manually do a G91X100 and then home, but computers are smart! Why can't they do it for me? 'especially since many of the users of this machine are *FAR* less technical than me, and won't know how to handle it, and call me. And it will likely happen every time the machine is turned on!

langwadt, 2016-08-09:
grbl only has one limit input per axis it could be at either or both end of travel so in the general case it has no idea what direction it is safe to move when it is at a limit switch

chamnit, 2016-08-09:
@emteeoh : Your 'problem' can be easily solved numerous ways.

First, you can set your G28 'home' position just outside of your limit switch, such that it's not being triggered constantly. Then Grbl won't throw a homing error when you try to home the next time. Easy. Solved. Done.

Second, it's a bad idea to put a motion command in a startup line. If you are in an emergency situation and happen to reset the machine, it'll execute that motion when you don't want it to. You should instead use a macro that most good Grbl GUIs have now days. Or, prefix/append your g-code programs to contain the startup and job completion g-codes for how you'd like your machine to behave. Most CAM programs allow you edit the post to add things like this.

Third, Grbl requires the limit switches to be disengaged prior to homing by design. If you have more than one switch wired to the same pin (a lot of people do for hard limits on opposite ends of travel), then there is no way to tell which one is triggered. The limit switches are also used as hard limits. Had you enabled this, you would have triggered a hard limit alarm every time your went to your G28 'home'. I don't plan on altering this behavior for homing. 

Fourth, if you'd rather have Grbl set the origin to where the homing cycle completes after the pull-off motion, you can uncomment the HOMING_FORCE_ORIGIN line in config.h, re-compile, and flash. The homing pull-off motion is a post-homing motion to move the machine off of the limit switch. Once that completes, it sets a 0,0,0 machine position. This would help you if you'd like clean numbers in terms of where your G28 is. 

Either way, this is a non-issue. This thread will be closed.

closed 2016-08-09 by chamnit

emteeoh, 2016-08-09:
@chamnit thanks for the detailed answer. Just a few comments:
re: first
Other positions either put the "tool head" in the way of loading and unloading, or out of reach, so its, unfortunately, not that simple.

re:second
Yep. I learned that lesson about 5 minutes after posting... I thought I had thought it through and it would be OK, but obviously, I was getting tired, and wrong.

re: third
And that explains the behaviour. Thanks.

re: fourth
Might do that. I'm not sure its necessary. I just have to decide if I'm going to modify GRBL to home properly for me, or just macros. If its left to me, I'll use the macros, but its not really my decision to make. Anyhow, thanks for the help!

chamnit, 2016-08-09:
@emteeoh : I think I'm failing to understand why your "tool head" can't be offset an additional 1mm to give a little more clearance to the limit switch, and how that would still be in the way of loading and unloading on a laser cutter. 

Anyhow, looking back at the code, my memory is apparently fuzzy. It looks like I altered the homing cycle to allow you to start it while the limit switch is engaged, so this issue is something different. To get this behavior, just make you have the LIMITS_TWO_SWITCHES_ON_AXES config option disabled, which is the default. 

In this case, Grbl will throw a homing fail if any of these circumstances occur: (1) A soft-reset or safety door ajar command has been sent. (2) The limit switch is still engaged after the pull-off motion. (3) The limit switch is was not found during the approach phase.

What it sounds like is you have an issue with (2). If you can, increase the pull-off motion value.

emteeoh, 2016-08-09:
The carriage design is such that the X limit switch is engaged anywhere from -75mm to 0. So its not a matter of moving "1 mm", its a matter of a third of a foot. I'll try 75mm on the pull off.