grbl/grbl #1052: Feed Override question

109JB, 2016-08-10:
Sonny, 
I believe that you are working on feed override for the V1.0 release of grbl, so if that is the case then this is a moot point and can be disregarded.

I finally have some free time to work on my gui and in doing so I have been thinking about feed override at the GUI level. I mashed up a test and it can be done, but the way I have it is it adds an F word in the next command sent if the feed override slider is changed in the gui. Any subsequent F words in G-code have the slider value, which is a range from 0.01 to 2.00 (1 to 200 %) applied as a multiplier.  The problem is that this only works for the next command to be sent and any already in the grbl queue have to run before the added/modified F word is received.

So, if you don't already have a method in place for this, can I suggest a new $ command, perhaps $F that is a feed override that is read immediately and applied as a scalar to the normal F commands in the G-code. I don't know the ins and outs of the grbl code since I don't know C programming that well, so maybe this is a juvenile approach that won't work, but it was a thought.

If this is already slated for implementation then please close this issue.

jahnj0584, 2016-08-10:
I dont know what Sonny has planned, but i have edited my feeds before with
notepad++ and used the replace option [search for F(value),replace with
F(value)] and you can then save a copy which will run faster or slower. Is
there a way grbl or UGS can automatically do this?

chamnit, 2016-08-10:
@109JB : Grbl v1.0's feed (and rapid) override execute in real-time and are commanded with a new set of real-time command characters that are in the extended ASCII space. There are about 15-16 new commands. Make sure your GUI can send extended ASCII like the other ~, !, ? commands. Of the select v1.0 testers, some experienced some issues with getting this to work in certain programming environments. I think it had to do with default character encoding issues.

These commands have a +/- coarse and fine adjustment, which default to 10% and 1% respectively. There is also a 100% reset command. Grbl will multiple the programmed feed rate by the feed override scaling factor and execute it within 50-60 ms. So no need to alter the F words in the g-code program itself with this new feature.

It's pretty awesome to get immediate feedback so you can dial in an optimal cut speed without risking overdoing it. It's simple to sneak up on it with the +/- 1% by just listening to the machine. The way people have been hacking override in with F word alterations has problems. You have to wait for the buffer to clear for way too long and risks breaking/wearing down a cutter or over-exerting the machine. 

Anyhow, going to close this issue as requested.

closed 2016-08-10 by chamnit

109JB, 2016-08-10:
Editing the g-code file cumbersome at best.  It is certainly something that needs to be done if a program is used over and over so that the next time it is run the feedrates are correct. I don't use UGS, but I know the current GRBL can't do this. I have been writing my own GUI, mainly to fit my own usage and as mentioned I did a test. I can alter the feedrate while the g-code is running, by injecting a new feedrate that is a multiplier of the existing feedrate, but this doesn't take effect immediately as it should.

The issue is that a user needs to be able to alter the feedrate while the machine is running and do this in real time. At this point, this is in my opinion the biggest downfall of grbl at this point in time.  When the machine is running a user needs to be able to slow down or speed up the machine for various reasons. For example a new program that has not been tested and in operation the user finds the feedrate is too fast or too slow. It could also be a case with an existing program where the cutter is maybe a little dull. Another case coud be a program written for a cutter with more or less cutting edges. For example the feedrate in  a program written for a 4 flute end mill would need to be cut in half for a 2 flute end mill.  These are just examples, but demonstrate how feedrate override is necessary to prevent the user from having to stop a run and then edit the code when an override would suffice.

109JB, 2016-08-10:
Sonny, you replied while I was typing. Glad to hear it is slated for V1.0. Can't wait to see it.

jahnj0584, 2016-08-10:
I agree with you 109, although I think there are some other issues that should be added before feed over ride, such as limit switch testing, backlash compensation, and stepper calibration plug in (I assume 2 of these actually depend on the GUI)

Maybe Im just used to using hobby/open source stuff, where some manual work to get stuff done is required and going off that. Its easier for me to just find out what my machine cuts well at, and always use those feeds + speeds compared to waiting for a feed override, although it would be a very nice luxury to have in the next few months.

chamnit, 2016-08-10:
@jahnj0584 : If I have the room (very small chance), I'm planning on adding backlash comp. Either way, it'll be installed on the Mega version soon after v1.0 is released.

jahnj0584, 2016-08-10:
Great. I am getting a new arduino anyways, what model should I do, that isnt the Uno? Im looking for something 32 bit or higher for better acceleration and pulse's

chamnit, 2016-08-10:
@jahnj0584 : A direct port of Grbl to a 32-bit platform will only give you higher pulse rates. Overall motion will not change much. I am currently working on a new 32-bit controller that will fully take advantage of the computing power, but I will not elaborate more than that.

tklus, 2016-08-10:
I know you are busy as we all are but I was curious on a rough timeframe for the v1.0 release.

Thanks!
Tim

chamnit, 2016-08-10:
@tklus : See my answer here. [https://github.com/grbl/grbl/issues/1046#issuecomment-237035359]

109JB, 2016-08-10:
Sonny,  can you explain a little more about how the feed override will work.  It sounds as if there are going to be something like 5 realtime commands +1%, -1%, +10%, -10%, and restore 100%.  

I am trying to think how I would implement this into my GUI.  Having come from LinuxCNC, which uses a slider and shorcut keys to implement feed override, I am trying to envision how this same approach could be implemented. For example in linuxCNC if you want to go from 100% to 10% feed override you simply press the "1" key and you are at 10%. The 2 key = 20%, 3=30%, etc.  

Say the feedrate is initially 100% and the user wants to go to 10%. It sounds like the new grbl implementation would have to go 100% - 90% - 80% ...10%.  Is this correct?  Not a big problem, the gui would just have to do some calculation and send multiple commands.

If the above is correct is there any reason that simply implementing the override with a single realtime command with a variaible, like $F=0.1  for 10% feedrate could not have been done? This would not be an eeprom saved command but injected at runtime?  Just curious.

chamnit, 2016-08-10:
@109JB : It's modeled after the Haas controller, such that override values can be altered via button presses, rather than a slider. It fit well with the real-time command character structure. Sure, I could add something to allow a specific value to be set by a $ string, but it won't be real-time. The command may have to wait in the serial RX buffer or if Grbl is waiting to finish queued commands before accepting new ones.

Even if I could make a '$' command execute in real-time, I'll say that I've thought about it and have received past requests for this same thing. However, I don't think adds much value for the flash cost or the additional release delays to develop it for v1.0. The individual commands should be fine. You don't really want to set a value too low too fast anyhow. In practice, it's arguable that it's better not to have the ability to over-compensate by setting a value.

That said, once v1.0 beta is out. There will be some room to make small adjustments to how it works with GUIs as a community discussion. But, I don't plan on developing the 328p version of Grbl past v1.0 since we've hit the flash wall hard. It'll be in maintenance mode only. So, keep that in mind. Most of these better/next-gen ideas are being pulled into the advanced ARM version, where the comm protocol will be a lot different and better suited to accommodate things like setting an override value in real-time.

109JB, 2016-08-10:
Understood.  Not a problem, mostly curiosity mainly.  As said, It can be slider implemented in the gui code anyway with just a few lines.

What are the chances of my getting on the list of testers for the pre-release version?