grbl/grbl #1053: Slaving Y axis/Driving 4 motors

jahnj0584, 2016-08-11:
Hi sonny 
I use 4 motors for my 3 axis cnc(two on Y) and have just ordered TB6600 digital drivers. It's not a good idea to drive two motors off one driver, so akin to the cncshield V3 I'd like to split the Y axis into 2 sets of Step and Dir outputs on the digital port row. Is there any way to do this, or are all pins already used for something? 
Will i just have to use pins 3 and 6 and have 2 wires coming out of each (one per motor driver)?

Thanks

chamnit, 2016-08-11:
@jahnj0584 : Wire two drivers to one axis output from Grbl. If needed, have one motor rotate the opposite way by swapping the leads on one of the windings (not between windings).

closed 2016-08-11 by chamnit

jahnj0584, 2016-08-11:
Got it. Just like how i have it now with my GShield.

Thanks