grbl/grbl #1054: Z shifting after 1000s of moves

auhopu, 2016-08-11:
I tried cutting 4 simple circular pockets and used jscut's ramping feature (i.e. instead of straight plunging, it ramps into the required Z - or creates helical movements in my case of circular pockets, to optimize tool life). The outcome is ~9k lines of gcode, half of which involve Z moves.

After running the gcode, I noticed that it went back to the given safe height of 3mm (Z WPos showing indeed 3), but it was actually about twice higher. Here's my train of thought to debug this.

- Wrong $102: This was my initial thought. However, a wrong $102 would only cause problems of scaling (i.e. if it was higher/lower, my Z moves would compress/expand proportionately). However, moving from Za to Zb and back to Za should go to the same Za, even if this is out of scale. In my case, Za shifts over time! Then I tried to manually give 30~40 Z moves and they were all consistent (and, btw, correct in scale too). Z0 would always go back and kiss the waistboard. So whatever happens, happens after 1000s of Z moves.

- Probing issue: I am probing in different locations on the waistboard and when going G1 Z0 it is spot on. No probing/zeroing issues.

- Skipped steps: This is the next area to question. However, no sign/sound of skipped steps. Furthermore, what makes me exclude this one is consistency. It would always shift the same. Even without load when "air cutting".

- Waistboard not level: Turns out also irrelevant. Even though I do plan to face mill the waistboard soon, I do my Z shifting comparison on the same X,Y (before running the gcode and in the end).

- Tool slipping: That would be a possible explanation. If for some reason the tool did not engage in the collet tight enough, and the cut was too aggressive, then theoretically the tool could slip upwards and present exactly what I experience (this is a good visualization to imagine what is actually happening). However, this is out of question too... as, again, upwards Z shifting eventually occurs (by the same amount) when "air cutting" !

I am left with two areas (that I can think of) as possible culprits.

- Buffering: If for some reason some gcode was taken out of stream, I would see Z shifts. However, the fact that I am cutting downwards and the shifting happens upwards, makes this less likely. Also, I am using bCNC which is supposed to be the (or at least one of the) best performers in streaming massive gcode.

- Rounding: This is the only one I have not debugged. I am using an ACME and therefore my $102 is ugly:
(200step/rev * 2microstep/step) / (25.4mm/in / 12rev/in) = 188.976microstep/mm

Which makes me wonder if 1000s of Z roundings are enough to cause Z shifting. One thing I thought to try today is to throw my $102 out of scale with a nice round number (say 200microstep/mm) and see if the same gcode still causes Z shifting.

Btw, I am using the X-Controller with GRBL version 1.0c.20151109 .

Any ideas, recommendations are very welcome.

chamnit, 2016-08-11:
@auhopu : First, I saw this post on at least 3 different forums. When you do that, it's difficult to tell which one is the appropriate one to respond to and also cause multiple conversations on the same topic, where one will often do. Please make sure you post to all of these forums to tell those communities the issue has been solved and link it to here.

To answer your question, Grbl itself is highly stable. It regularly executes g-code programs on the order of millions of lines of g-code (3d carving). That said, I'm starting to find more issues related to Z-axis lost position due to helical plunges and mis-configured Z-axis accelerations. In these cases, the CNC machine can slowly lose steps due to the weight of the Z-axis and spindle itself. So, it'll lose steps even when it's not cutting, because it's still under load. 

The solution is to lower the Z-axis acceleration. Start by HALF of the current value. I've found that most of these past instances the acceleration was set way too high. Also, make sure your 'junction deviation' setting is around 0.01mm. I should also add that, in a recent X-Controller build, I updated some code to help mitigate this issue with a slightly more accurate model to handle these helical plunges, but altering the acceleration has still proven to be the most effective solution.

closed 2016-08-11 by chamnit

auhopu, 2016-08-11:
@chamnit thanks for the fast response and apologies if posting in multiple forums caused confusion. I was just running out of ideas as of what to test next.

I will point the rest of the discussions to your reply here. I will also further experiment with $11, $120, $121, $122 values.

jahnj0584, 2016-08-11:
if it helps my Z accel is 150mm (6 inches).

Put a sharpie mark on your coupler, and a matching one on your motor's
shaft. After the program, make sure they are still in line with each other.
This shows if its loose or not.

auhopu, 2016-08-11:
My defaults were:
$11=0.02
$120=500
$121=500
$122=50

and by looking at GRBL's [defaults.h](https://github.com/grbl/grbl/blob/edge/grbl/defaults.h) for the 500mm X-Carve
`#define DEFAULT_X_ACCELERATION (500.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`
`#define DEFAULT_Y_ACCELERATION (500.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`
`#define DEFAULT_Z_ACCELERATION (50.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`

I'm confused as to whether they should be 500-500-50 or 25-25-25 as suggested by the comments.

jahnj0584, 2016-08-11:
I have
$110=3750 (x max rate, mm/min)
$111=3250(y max rate, mm/min)
$112=5000 (z max rate, mm/min)

$120=75 (x accel, mm/sec^2)
$121=40 (y accel, mm/sec^2)
$122=150 (z accel, mm/sec^2)

Dont know how you drive your machine, but thats what works best for my ball
screw machine.

chamnit, 2016-08-11:
@auhopu : The helical plunge issue is made worse by a large difference between the X & Y relative to the Z axis. You can also try reducing these as well. Also, try getting a feel for how much friction there is in the Z-axis throughout travel. A tight set of rollers or a mis-aligned leadscrew can cause binding is certain areas.