grbl/grbl #1055: Non-linear transformation

soswow, 2016-08-14:
Hello,
I have a vertical plotter, that hangs on two wires. It changing it's location by changing length of each wire winding it on a spool.
The coordinate transformation from XY to AB is non-linear:
```javascript
  const y2 = h - y;
  const x2 = w/2 - x;
  const aAbs = Math.sqrt(Math.pow(y2, 2) + Math.pow(x2, 2));
  const bAbs = Math.sqrt(Math.pow(y2, 2) + Math.pow(w - x2, 2));
```

Currently I solve problem, by 
1) Preparing SVG where I break all the arc and lines into small straight segments
2) Generate normal XY Gcode
3) Run through all G00 and G01 commands and modify each point with that transformation

What I would really like is to have all transformation done inside firmware, so that I don't need to cut SVG into peaces and send all the possible G commands. 

I've seen similar question asked here long time ago: https://github.com/grbl/grbl/issues/33 but repo is dead.

I've looked into code and thought that if I introduce additional type of kinematics like CoreXY and add all my things in all the places where COREXY is used. Main place would be in `planner.c:plan_buffer_line` as I understand?

+ I have a fear that non-linear operations like SQRT/power are not gonna work in rapid cycles inside firmware, since it should be as light as possible.

soswow, 2016-08-14:
Ok. After initial investigation (which still is going) it looks like I will have to creation something similar to `mc_arc` even for linear motion commands. Since I want to end up with small chunks same as `mc_arc` does.

soswow, 2016-08-15:
I need 2 things:

1)  Modify `plan_buffer_line` where I need to transform XY coordinates into AB ones
2) If given line segment is too long in AB space, I would need to brake this line into smaller line segments (in XY space) and then pass it through 1) again.

Am I thinking right?

sTVG, 2016-10-23:
I wish i knew,
I've been researching Grbl for polargraph/drawbot/v-plotter and can't seem to find definitive answers. If someone who understands GRBL well can advise that would be very helpful!

chamnit, 2016-10-23:
@sTVG : Definitive answers for what? It's not clear what you are asking. If you are asking if Grbl supports polar coordinates, no it does not.

closed 2016-10-23 by chamnit

winder, 2016-10-23:
@soswow @sTVG Here is a fork which added polar coordinates, maybe it can help you:
https://github.com/ilaro-org/grbl-polar

soswow, 2016-10-23:
Those are not polar coordinates, technically.

@chamnit What @sTVG  was asking is not body of this issue. I opened this issue with hope of getting some advice on 1) if you think it's possible at all (sqrt/power operations) and 2) if my thinking on where it should be placed is correct. More details in main text. Unfortunatly after 2 months I didn't get any response and issue is closed now =(

@winder thank you! Will look into that.

chamnit, 2016-10-23:
@soswow : Then polar bots shouldn't be called "polar" if they don't use polar coordinates. :) Either way, you are doing a coordinate transformation of some sort that Grbl doesn't explicitly support.

1) Yes sqrt and pow are possible, but depends on how many. Floating point is very expensive on an 8-bit AVR. You can probably get away with adding about 1-3ms of time to parsing and planning a g-code block. Maybe more if you are moving slow.
2) Yes, the transformations would be properly place wherever there is CoreXY code. You are doing the same type of thing.

Sorry, I try to respond to everyone but I get a lot of questions, all of the time. If an issue hasn't been answered after more than a month, I assume the issue is closed because, more often than not, the person asking has solved their own problem. It's a giant waste of time answering an issue that isn't useful anymore. It's hard to tell if you haven't unless you keep the issue alive by asking again after an appropriate amount of time (like greater than a few days.).

sTVG, 2016-10-24:
Great to see some communication happening here. So i was wanting to find out roughly the same thing. Basically doing what @chamnit did with COREXY for Polar. I have already tried the https://github.com/ilaro-org/grbl-polar, which has some polar transformation in the motion_control.c but it wasnt working properly, and i tried for 4 days to get it to plot. 

So my question is:
1. can Arduino (ATmega328P) handle the square root on the floating calculation for movement? it seems like it freezes some times
2. how would the steps per millimeter be set (because its a moving target - non linear transformation)
3. the ilaro-org/grbl-polar looks like it just needs a few tweaks, or maybe just some documentation (and yes it would be nice if they could explain it themselves, but they are spanish, and haven't, so any insight on how to make a transformation work would be very helpful)

thanks

soswow, 2016-10-24:
@sTVG 
2. Steps per mill doesn't change, since you know you spool C and it doesn't change. Machine is not thinking in X and Y, but in A and B. A and B are string lengths. And steps per mill for strings doesn't change (if you don't take into account when spool is full on first layer and C changes a little bit)

I will look into their solution more thoroughly. From I what I've seen so far this might be something I was looking for. It should be cleaned, though.

winder, 2016-10-25:
Maybe This fork of grbl is useful for what you guys are trying to do:
https://github.com/ilaro-org/grbl-polar

sTVG, 2016-10-25:
@winder, do you understand this fork? if so please help fill me in. Does it only work with absolute co-ordinates because it has to calculate everything relative to its known origin?

soswow, 2016-10-25:
@winder but you already suggested that above, no?
@sTVG not sure I understand your question. Everything is relative to it's known position, in all cnc machines. Usually this known position is origin, that you get by homing. In case of this branch they just hardcoded position and then manually position head at this point. I do the same with my drawing machine.

winder, 2016-10-25:
Sorry I'm not sure why that got posted a second time. I'm not familiar with the fork, just remembered it being shared a while back.

Chicodarave, 2018-05-12:
Trying this in 2018... Cant find a solution :(