grbl/grbl #1056: How to set up Door override?

PocketNC, 2016-08-15:
Hello Grbl users, 

Does anyone know how to wire / setup the door override function? I did not see anything about that on the wiki.

chamnit, 2016-08-15:
@PocketNC : There isn't  door override function in Grbl, but rather a safety ajar feature. I haven't written anything on it in the wiki, since this isn't something used by normal users. Mainly OEMs that need to ship a safety door for international sales. 

To enable it, you need to alter the config.h to use the feed hold pin as the safety door pin input. Wire your safety door switch to that pin. 

In addition, Grbl has a software command that uses the '@' character to enact the safety door code. That is enabled by default. And, a configurable parking motion that works like a feed hold but includes de-energizing and retracting the spindle. This is not enabled by default and only exists in the edge branch and Grbl-Mega code bases. This is due to the 1-1.5kB of flash space it requires to use it.

closed 2016-08-15 by chamnit

PocketNC, 2016-08-15:
Thanks!
We are a OEM working on the FR4 Machine shield at Pocket NC.  Also, how do we support GRBL like that of carbide 3d or inventables?

Feel free to email me direct,  info@pocketnc.com
-Matt