grbl/grbl #1057: stepper configuration help

nima5100, 2016-08-17:
hello every one.
sorry for not well english writing

i have some problem in configuration of grbl setting.my homemade cnc made by cnc shield and grbl . stepper drivers are A4988 .
i use belt and pulley and for x and y directions in one rotation of stepper belt run 92 mm.
for x direction i use 1.8 deg FH6-1570 01 (TYPE 103H7121-1741) sanyo denki stepper
and
grbl setting sets : $100=2.000 (x, step/mm) $110=10000.000 (x max rate, mm/min) $120=20.000 (x accel, mm/sec^2)
$130=200.000 (x max travel, mm)
x stepper runs very
but stepper runs very slowly and not continuous. is there any problems or stepper runs well?
https://youtu.be/CKYF6bF2h_I

langwadt, 2016-08-17:
sure that you have the wiring to the stepper motor correct?

jahnj0584, 2016-08-17:
did you calibrate the a4988's like the video said with Vref? It also looks like its in whole or single step mode. You need to put 2 jumpers under each driver.

nima5100, 2016-08-18:
@langwadt    I dont know. please help me about this problem.
@jahnj0584    NO. what video? please share video link to me.

jahnj0584, 2016-08-18:
https://youtu.be/89BHS9hfSUk

closed 2016-09-03 by chamnit