grbl/grbl #1058: Reading Arc Move errors

DOUG888, 2016-08-17:
With programs like BobCadCam the initial noting of the G02 (2 or 3) in arc lines is not repeated every line of code but is only added after the incidence of a different G01,G02/G03, this seems to cause an error and stops Universal Gcode Sender (latest version).

see an example below
N21 starts with a G02 which is reflected thru lines N22-N26 where it changes to N27 G01 then N28 G02 again. Where ever there is a missing G02/G03 on an I J line there is an error and UGS stops.

I have run script where the G code is in front of every I J line and it work perfectly.

I am using latest versions of GRBL and UGS Platform on a Windows7 machine

I am not sure if this is a GRBL issue or a UGS issue.
Thanks

N20 G01 X200.587 Y71.96
N21 G02 X200.392 Y73.038 I4.107 J1.3
N22 X200.912 Y75.04 I3.432 J.177
N23 X201.729 Y75.979 I3.093 J-1.866
N24 X201.98 Y76.163 I2.027 J-2.499
N25 X203.664 Y76.68 I1.718 J-2.594
N26 X204.824 Y76.536 I.047 J-4.368
N27 G01 X204.828 Y76.535
N28 G02 X206.794 Y74.417 I-.841 J-2.753
N29 G01 X206.795 Y74.413
N30 X206.838 Y74.175
N31 G02 X206.823 Y73.442 I-1.819 J-.328
N32 X206.485 Y72.908 I-.918 J.206

chamnit, 2016-08-18:
@DOUG888 : Grbl honors the modal motion modes like G2 and G3, so they should be accepted and executed by Grbl. 

If Grbl is operating correctly when you stream it via some other GUI or streamer script, then it's not a Grbl issue. Some GUIs like UGS do a little bit of pre-processing that include rounding values to a certain precision and bCNC that checks for valid g-code commands internally. At times, these things can have a bug and prevent your program from operating correctly. I would bring this issue up on UGS' github website.

109JB, 2016-08-18:
I tested your code snippet in my streaming program and no errors were produced. It must be a UGS problem.

DOUG888, 2016-08-21:
I am posting this for anyone who has had a similar problem.

I had been having error problems when using Arcs. If i had an Arc line with no G code following an Arc line with G02 or G03 that line would cause an error. This would happen if a G00 line was between the G02/G03 Arc line and other Arc lines, The G00 would cancel the flow on effect of the G02/G03. Frustrating to say the least, this also happened when using R radius instead....the solution was in the POST, i modified the line below to a "n" value which gave me G code on every G02/G03 required line, now runs perfectly.
I am using Bobcam V24 on Windows7.

"204. Are the g codes (G02 and G03) modal in arc milling? n <<<<<------------- this value was y"

Original Code
N63 G02 X22.124 Y66.79 I2.119 J8.237
N64 G00 X28 Y69
N65 X33.696 Y72.598 I8.317 J-2.139 <<<<<------------------------error was here and next line
N66 X74.864 Y49.249 I-80.481 J-189.872

New code after POST was changed
N54 G03 X28.68 Y49.821 I.002 J3.348
N55 G02 X30.167 Y49.803 I.691 J-4.436
N56 G02 X31.328 Y49.512 I-1.563 J-8.681
N57 G01 X32.2 Y49.229
N58 G03 X33.001 Y49.006 I2.971 J9.152
N59 G03 X33.766 Y48.894 I1.068 J4.612

Not the most elegant but successful.

chamnit, 2016-08-22:
@DOUG888 : Your original g-code is not correct. All motion modes are modal, including G2 and G3. At N63, you are in G2 arc mode. In N64, you are in G0 rapids mode. In N65 and N66, you are STILL in G0 rapids mode. Rapids mode does not accept the IJK words. You need to specify the arc mode you need at the very first instance it occurs. Each following motion command will assume it was the last mode set.

Please read up on g-code modal states. LinuxCNC has some very good online resources on this matter.

closed 2016-08-22 by chamnit

109JB, 2016-08-23:
As Sonny said G2/G3 are modal, but so are G1 and G0 and all of these codes are in Group 1 motion codes. Only one Group 1 code is active at any one time. So using a G0 or a G1 cancels G2 or G3 until it is specified again. Only the first instance in successive moves has to have the G2 or G3 code.

DOUG888, 2016-08-23:
Yes!..I agree with both comments..the G00/G01 cancels out the preceding G02/G03. 

> The G00 would cancel the flow on effect of the G02/G03. 

I did read up the on the forums and thats where i got the hint on this

The only way I could fix this was to do what I said above and that was to modifie the POST. If there is another way. please tell me and i will try it, if not, this has fixed the problem for me.