grbl/grbl #1060: X-Axis not homing

dattasaurabh82, 2016-08-18:
So I've cloned the latest grbl 
```
>>> $I
[0.9j.20160317:]
```

I'm using simple lever push switch at the furthest point of z, y and x axis. They are in default normally high. And I checked 
`$5=0`

and my **soft limit, hard limits** are off and **homing** is on
```
$20=0
$21=0
$22=1
```
Checked the switches with multimeter for conductivity test . When I press them they are closed. All of them . 
X axis Limit switch to pin 9 of Arduino
Y axis Limit switch to pin 10 of Arduino
and Z axis limit switch to pin 12 of Arduino. 

So all the connections are okay as well. 

Now when I do homing,
 Z axis is homed and the coordinate get's set to 0 in Universal G-Code sender UI.
Then the Y axis is homed and it's coordinate get's set to 0 in UGS UI as well.
But the x Axis is not homed . bUt I can jog x-axis at any point of time and it works correctly when I send it a a g-code for motion. 

So then I checked the `config.h` file. 
There I found 
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
```
Then I made some changes, just to see, likewise
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 (1<<X_AXIS)  // OPTIONAL: Then move X,Y at the same time.
#define HOMING_CYCLE_2 (1<<Y_AXIS)                        // OPTIONAL: Uncomment and add axes mask to enable
```

But still the same sequence i.ce x axis not being homed. 
Sometimes after homing in the UGS  UI, it shows x = 0 along side z=0 and y=0, whereas the laters are being homed and x axis is not. 

Where are the actual homing sequence being written (in which file)? It seems the X-axis homing is not being executed. Seems like a piece of code is missing as there is not delay after z axis homing and it goes to home Y axis. But in between there is X axis in the config.h (that's where I got the idea of sequence). 

So is it a bug can or you help me solve it?

On separate note::
How can I auto-magically enable Homing on machine start . 
I tried `$N0=$H` with no luck obviously as it seems `$N` s only execute G-Code parsed modes.
what is the parsed mode for Homing in terms of `G` s

109JB, 2016-08-18:
Homing is an issue that comes up often and the homing in GRBL is proven to work. I myself have no issues with homing. Most of the time the issue is with electrical noise that causes the pin state to change. This makes GRBL think the switch was activated.  First step is to check wiring and use good practices to guard against unwanted electrical noise. If that doesn't work, you can try changing the debounce setting. If that still doesn't work, you can add a small capacitor to the pin to filter the noise.

In my opinion, homing at power up is a bad idea. When flipping the power switch on the machine I don't want it doing something on its own accord. Also, if homing is implemented in this way you would have to code it so grbl only does it on a hard reset, as a soft reset would run the $N codes.  Soft resets have been used by interfaces to achieve other things tha may not be obvious. You would not want a soft reset initiated by the interface to cause your machine to suddenly home during a program run. What are you saving anyway. It only takes one button push to get the machine to home.

dattasaurabh82, 2016-08-19:
Solved. Was the classical problem of interference. So put a filter cap 100nF  between Digital Pin 8 and GND and it worked.

closed 2016-09-03 by chamnit