grbl/grbl #1061: Max value for X, Y & Z step/mm (v0.8c)

manusnz, 2016-08-18:
Hi there,

I'm wondering if anyone knows what the max value for the above ($0, $1 & $2) is? I seem to come up with a very high number (2285) when I run the calcs for the x, step/mm variable.

When I calculate using the "steps_per_mm = (steps_per_revolution*microsteps)/mm_per_rev" to work out how many steps my stepper must turn to move my lead screw 1mm I end up with 2285.714 - because I have pulleys involved, this number will be well out (see below for pulley details)

Thanks...

Feel free to skip this bit, but it gives more insight to my setup:
To further complicate things I have the stepper driving a belt via two pulleys (stepper has a 12t XL and the lead screw a 30t XL) so I've got a ratio of 2.5:1 and one full rotation of the lead screw moves the bed ~1.4mm

So even if I follow this and break it down to how many (micro?)steps to get 1mm of bed movement, I end up with ~5714 (micro)steps! 
200(steps) * 16(micro) * ( (1/1.4) * 2.5)

If I reverse engineer the number I need to enter into universal GCode Sender to move the bed 1mm, I use the following command: G01X17.2 (I got this through trial and error)
Running that through that I come out with 4300 steps (17.2 * 250 - 250 is the default steps/mm)

swarfer, 2016-08-18:
Hi
The steps/mm values can be large since they are floating point numbers.

Instead of trial and error, what about some simple math?  (-:
http://www.openbuilds.com/threads/c-beam%E2%84%A2-machine-plate-maker.2349/page-36#post-34484

An initial calculation might be
(200*16*30/12) / leadscrewpitch (you said about 1.4mm)
(200*16*30/12) is the number of steps to turn the leadscrew one revolution
ergo
8000/1.4 = 5714.2857 steps/mm

Now do the calibration process at the web site I gave.   
Use as many digits as you can get from your calculator, it does make a difference on long travels.

If you want more speed and power from your system, and are willing to give up a little resolution, you can set your drivers to 8x microstepping (or even 4x).  This gives ~2857.1 steps/mm but more power.

Using $0 $1 and $3 means you are using GRBL 0.8 - any particular reason for that?  0.9 is much better (-:

jahnj0584, 2016-08-19:
Theres no reason to go below 1/4 or 1/8 usteps. I have 320steps/mm with my 1605 ballscrews.

Theres no difference in set up from 0.8 to 0.9, the $ symbols carry over.

swarfer, 2016-08-19:
"Theres no difference in set up from 0.8 to 0.9, the $ symbols carry over. "
really?  not according to the wiki
for 0.9 
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
...

for 0.8
$0=755.906 (x, step/mm)
$1=755.906 (y, step/mm)
$2=755.906 (z, step/mm)
$3=30 (step pulse, usec)
...

jahnj0584, 2016-08-19:
0 1 and 2 just become 100 101 and 102. If you're upgrading anyways David,
just skip the step and work with the current version.

closed 2016-09-03 by chamnit