grbl/grbl #1062: Homing stalls, resulting in a Alarm after reset

Eriond, 2016-08-19:
I have read through a lot of posts regarding homing issues, and I'm quite confident that GRBL handles homing perfectly. Unfortunately, not in my case. I have a set-up for a laser engraver (X and Y axis, no Z) which works perfectly fine when jogging around. I added limit switches at both ends (in parallel) of X and Y axis. They are now also RC-filtered according to the instructions in (http://www.eng.utah.edu/%7Ecs5780/debouncing.pdf). The limit switches do their job when i jog into them manually.
My issue is that when I send a $H (UGS 1.0.9), the machine activates (but doesn't move) the motors. I can hear them hum slightly, but no movement... indefinitely. If I clear the situation with a "Soft reset" I get a "ALARM: Homing fail, ok" response and the motor humming stops.
If it is a clue, I get the same motor humming without motion during the upload sequence when flashing with Arduino (1.6.11).

jahnj0584, 2016-08-19:
is your homing seek speed too slow?

gitBugs, 2016-08-19:
Did you remove the Z-axis homing from config.h before you compiled GRBL?

By default, GRBL homes the Z-axis first, then homes X&Y. If you have no Z-axis but leave it on this setting, you see exactly what you've described: motors engage, but sit there forever (well, until GRBL gives up with "homing fail") while GRBL waits for a Z-limit switch that never comes.

Quick way to test this is to send $H, then manually short the Z-limit pins a few times, as if a limit switch is being tripped and released during the Z-axis homing process. After a few goes, GRBL should be satisfied that Z is homed then move on to X&Y.

The fix is described here: https://github.com/grbl/grbl/issues/224#issuecomment-226196204

Eriond, 2016-08-19:
@jahnj0584 : Homing seek = 1000mm/s, Homing feed = 25mm/s
@gitBugs : This is what I've done to the config file:
`// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
// #define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
// #define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)`

I will try your suggestion with the Z-limit input though, and report back.

Eriond, 2016-08-19:
The result is remarkable. When toggling pin D12 a couple of times, the homing cycle began!
Not only did it begin, but it did so using X and Y axis simultaneously, something I clearly changed in the config.h file (see post above).
I'm beginning to suspect that Arduino IDE ignores my changes (or the other way around; that I'm stupid and making changes in the wrong file).

jahnj0584, 2016-08-19:
You have to re-flash your arduino with the Xloader/new grbl everytime you
make changes

Eriond, 2016-08-19:
I thought that was done each time i clicked "Upload" in the IDE. But now I'm not sure where the IDE looks for the config.h file...

jahnj0584, 2016-08-19:
The config is for grbl. You have to reflash (with xloader) anytime you
recompile the source code (which is what you do in the IDE)

Eriond, 2016-08-19:
Now we're a bit off topic, but when the IDE says "uploading" it doesn't actually update the firmware on the Arduino?! Whats with all the flashing LEDs then?
This is my first experience with Arduino and AVR, I have only worked with PICs before.
Ok, suppose I need to use xloader. Where do I find the compiled binary?

chamnit, 2016-08-19:
You need to alter the config.h file inside the Arduino library folder. It copies Grbl's source code there.

closed 2016-08-19 by chamnit