grbl/grbl #1063: Can't get Ubuntu 16.04.11 to load grbl

tjfisher26, 2016-08-20:
Having trouble loading library and the ability to flash grbl to my arduino . Please help

closed 2016-09-03 by chamnit

tjfisher26, 2016-09-03:
Thanks for nothing

chamnit, 2016-09-03:
If you want and expect help from others, please be nice and explain in detail what the problem is. A dozen or so words doesn't cut it.

tjfisher26, 2016-09-04:
Closing a question with even trying to help yourself if flat out rude so no point and trying

chamnit, 2016-09-04:
@tjfisher26 : I'm sorry that I closed your issues thread prematurely. It's been open for two weeks without anyone volunteering their free time to help you, which I did not personally have. Typically at this point, the person asking has already resolved their problem on their own, or mentions that their issue is still outstanding when it get closed. Usually in a more diplomatic way. At that point, I'm happy to re-open it.

But, your issue post consists of a very vague and open ended question, which often takes much more work on the helper's side, than the questioner's. Someone has to take the time to go back and forth to ask you want exactly is wrong to help you diagnose your problem. The effort goes both ways my friend. 

If you still need help, I'll be happy to help however I can, when I have the free time to volunteer for your issue.

tjfisher26, 2016-09-04:
We can pass on re opening it . I pulled the Hard drive and replaced it with a Windows drive . I was hoping to have a Linux base system to run my mill but I will just deal with Windows 

Thanks

funlw65, 2016-09-04:
Usually, when you install Arduino from Ubuntu repository, the installer will set your system to detect your Arduino after a reboot. If it doesn't work from the first try, then your system needs tweaks but no guaranties that it will work in the end. So, you did well installing a Windows machine where you can (and know how to) make it work. Making a connection between a Linux system and a microcontroller board using USB when this connection didn't worked from Arduino IDE install, it is a job for the advanced linux user, not beginners. Anyway, the best place for such a question is Arduino forum or Ubuntu forum.