grbl/grbl #1068: Maybe issue with GRBL?

Piero87, 2016-08-27:
Hi,
I just finished my CNC, and my first project is using a pen to test the configuration of the CNC, i have 4 Nema 23, a GT3 6mm Belt, GT3 1/4 20 teeth pulley and an Arduino Clone + a CNC Shield v3.50 with last version of GRBL.
I have change some configuration of grbl, I haven't touched some other configuration because I have understand it, this is my config now:

$0=3 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=26.670 (x, step/mm)
$101=26.670 (y, step/mm)
$102=200.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=150.000 (x accel, mm/sec^2)
$121=150.000 (y accel, mm/sec^2)
$122=150.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=290.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

This is the Data sheet of my Nema 23:

[img][url]https://www.dropbox.com/s/lrl2zjkq0afef8k/IMG_0765.JPG?dl=0[/url][/img]

If anyone can suggest me some settings i will appreciate. Anyway i'm explain here my problem, this is the svg i tried to draw with Easel:

[img][url]https://www.dropbox.com/s/km977sy47e3lz27/Schermata%202016-08-26%20alle%2021.57.40.png?dl=0[/url][/img]

This is the first result:

[img][url]https://www.dropbox.com/s/t28sesbrjwhkuca/IMG_0766.JPG?dl=0[/url][/img]

As you can see when finished to draw the R of IRON it should go the Z axis up and move to the top to draw the tie, instead the Z axis go further down instead of go up and draw a line until the Tie. I can't understand why. 
I have tried to draw it again, with the same setting and this is the result:

[img][url]https://www.dropbox.com/s/b3vq5l7c6rt4qyd/IMG_0767.JPG?dl=0[/url][/img]

as you can see it was a problem again in the written IRON MAN, this time doesn't go down with the Z axis but the letters are not the right shape. Someone can tell me what could be the problem?
If i use the machine control buttons of the Universal GCode Sender to move the CNC left right up down, everything seems to be fine, but when i run a test that is the result.

I hope someone can help me.

Thank you very much!!!

jahnj0584, 2016-08-27:
> 0765.JPG?dl=0[/url][/img]
> Schermata%202016-08-26%20alle%2021.57.40.png?dl=0[/url][/img]
> 0766.JPG?dl=0[/url][/img]
> 0767.JPG?dl=0[/url][/img]

Did you calibrate the voltage of the driver's?

Piero87, 2016-08-27:
No I haven't touched that voltage, I must do it?
Anyway I have tried with the software UniversalGCodeSender and works very well with no problem, it can be the EASEL software to do that problem? i have make several test with UniversalGCodeSender and without one failure

X3msnake, 2016-08-27:
Yes it can.

Maybe it is shipping some gcode.

You should check the ref voltagem even not related to this problem just to
make sure you are within the range otherwise in the long run you risk
burning a driver or a motor.

Low power to the driver may cause the motor to skip steps and sometimes
lose grip wich can cause failed features.

I assume your gcode on easel and ugcs ia the same. If not check if the
failure part is not a fast movement part that may cause the motor skipping

X3msnake, 2016-08-27:
"Skipping not shipping" sorry

No dia sábado, 27 de agosto de 2016, Vinicius Silva <x3msnake@gmail.com>

Piero87, 2016-08-27:
Thanks for the answer, how I can check the ref voltage?

X3msnake, 2016-08-27:
Mesure the voltage between the screw and the ground pin

And ajust acording your motor amp specs on the datasheet

The fórmula for the polulu type drivers is A = 2xrefV

Piero87, 2016-08-27:
My Driver is DRV8825 is the same?
This is the data sheet of my Nema 23: 
https://www.dropbox.com/s/lrl2zjkq0afef8k/IMG_0765.JPG?dl=0

What voltage I have to set?
can I get the ground pin of my CNC Shield? or the ground the each driver?

X3msnake, 2016-08-27:
Ok so you motor is rated at 2.8A, therefore the ref V should mesure 1.4V on
each driver.

If all is wired correctly the ground should be common, so yes you can use
any ground. Just for the sake of sureness mesure once with the ground of
the drv8825 aka polulu driver and once with the shield ground. If there are
no diferentes in voltage then you can safely assume all is grounded
correctly into a common ground and you can use the cnc shield ground.

The drivers in the pic are the same as yours the ground pin is the first on
on the bottom right while the ref ajust screw is facing upwards.

Just check the drivers datasheet in case of doubt

https://www.pololu.com/product/2132

The ajust screw is very sensitive you need only small ajustment to Adjust
the ref voltage.

jahnj0584, 2016-08-27:
https://www.youtube.com/watch?v=89BHS9hfSUk

lalo-uy, 2016-08-27:
Check this

http://reprap.org/wiki/Pololu_stepper_driver_board

chamnit, 2016-08-27:
@Piero87 : It's highly likely you have your Z-axis acceleration set too high. Try lowering by half first to see if it gets rid of your problem.

closed 2016-09-03 by chamnit