grbl/grbl #1069: Probing error ("Bad number format")

dougkusanagi, 2016-08-27:
I'm trying to make a gcode sender in PHP and testing some sends i got that error just with some numbers...

This is what i received:

**Enviado: G0x50.0 
Recebido:
ok

Enviado: G0x40.0 
Recebido:
error: Bad number format

Enviado: G0x50.3 
Recebido:
ok**

That's my code:

```
<?php

// echo chr(10);

include "PHP-Serial/src/PhpSerial.php";

exec("mode COM8 BAUD=57600 PARITY=N data=8 stop=1 xon=off");

$serial = new phpSerial();

$serial->deviceSet("COM8");

$serial->confBaudRate(57600);		//Baud rate: 9600
$serial->confParity("none");			//Parity (this is the "N" in "8-N-1")
$serial->confCharacterLength(8);		//Character length     (this is the "8" in "8-N-1")
$serial->confStopBits(1);				//Stop bits (this is the "1" in "8-N-1")
$serial->confFlowControl("none");

$serial->deviceOpen();
sleep(3);

if (ob_get_level() == 0) ob_start();			//Inicia o Buffer

$read = $serial->readPort();
echo "<pre>".$read."</pre>";

ob_flush();							//Start Buffer
flush();								//Send Buff

$gcode[] = "G0x50.0\n";
$gcode[] = "G0x40.0\n";
$gcode[] = "G0x50.3\n";

foreach ($gcode as $line) {
	echo "Enviado: " . $line . "</br>";
	$serial->sendMessage($line);

	$read = $serial->readPort();
	echo "Recebido: <pre>" . $read . "</pre>";

	ob_flush();
    flush();

	if(!$read == "ok"){
		$serial->deviceClose();
		die($read);
	}
}
```

chamnit, 2016-08-27:
Grbl's serial port rate is 115200.

dougkusanagi, 2016-08-27:
My GRBL is not the new version, so baud rate is 57600 tnx for answer.
:)

electrokean, 2016-08-27:
This is almost certainly a parity error.
From the fact you're calling the `mode` command, I assume you're running on Windows.
I seem to recall on Windows the phpSerial parity setting doesn't work properly - each time you change a setting Windows defaults all the others.
Try removing the lines to set baud rate, parity, etc, (i.e. between `deviceSet` and `deviceOpen`) and just rely on the `mode` command and see if that works.
Of course, if you still have problems and expect assistance here, you should really test with the latest version.

dougkusanagi, 2016-08-29:
I tryed comment set baud rate, etc in serial php class...

`include "PHP-Serial/src/PhpSerial.php";

$serial = new phpSerial();

$serial->deviceSet("COM8");
exec("mode COM8 BAUD=57600 PARITY=N data=8 stop=1 xon=off");

// $serial->confBaudRate(57600);			//Baud rate: 9600
// $serial->confParity("none");			//Parity (this is the "N" in "8-N-1")
// $serial->confCharacterLength(8);		//Character length     (this is the "8" in "8-N-1")
// $serial->confStopBits(1);				//Stop bits (this is the "1" in "8-N-1")
// $serial->confFlowControl("none");

$serial->deviceOpen();`

didn't work...

electrokean, 2016-08-29:
> didn't work...

If you want assistance, you should provide a bit more info than that.  What output did you get this time?  Exactly the same, or different?

I'm pretty sure this issue is just due to a bug in phpSerial on WIndows as the errors you got would be consistent with incorrect parity (i.e. other than "none").  There is actually an uncommited patch to phpSerial from 2 years ago for this issue, so it looks pretty well unsupported.  See https://github.com/Xowap/PHP-Serial/pull/10

Maybe you could try that patch, or just other combinations of commenting/uncommenting or ordering of those lines.  In particular, I'd try moving the `confParity` call to the end of the list.

Or... try it on Linux, using a different serial library, or another language - Python is good :)

dougkusanagi, 2016-08-30:
I tryed upload other GRBL version, grbl0.9f precisely, config the mode manually and remove the phpSerial from the code and surprisely worked!
But, now i send more lines of gcode and starting of line 3, the arduino stop send me back the response, and just return nothing...

`<?php

include "PHP-Serial/src/PhpSerial.php";

exec("mode COM8 BAUD=115200 PARITY=N data=8 stop=1 xon=off octs=off rts=on");

@$fp = fopen ("COM8", "r+");
if (!$fp) {
   die("Porta não encontrada ou indisponível.");
} else {
   echo "Open";
}

sleep(3);

if (ob_get_level() == 0) ob_start();	//Inicia o Buffer

$read = fread($fp, 1024*8);
echo "<pre>".$read."</pre>";

$gcode[] = "$$\n";
$gcode[] = "g0x50\n";
$gcode[] = "x50\n";
$gcode[] = "x50\n";

foreach ($gcode as $line) {
	echo "Enviado: " . $line . "</br>";
	fwrite($fp, $line);

	$read = fread($fp, 1024*8);
	echo "Recebido: <pre>" . $read . "</pre>";
	// var_dump($read);

	ob_flush();
    flush();

	// if(!$read == "ok"){
	// 	$serial->deviceClose();
	// 	die($read);
	// }
}`

The response:

Open
Grbl 0.9f ['$' for help]
������������������������������error: Expected command letter
Enviado: $$ 
Recebido:
$0=200.000 (x, step/mm)
$1=755.906 (y, step/mm)
$2=755.906 (z, step/mm)
$3=755.906 (x max rate, mm/min)
$4=0.000 (y max rate, mm/min)
$5=0.000 (z max rate, mm/min)
$6=0.000 (x accel, mm/sec^2)
$7=0.000 (y accel, mm/sec^2)
$8=50.000 (z accel, mm/sec^2)
$9=-0.050 (x max travel, mm)
$10=200.001 (y max travel, mm)
$11=200.000 (z max travel, mm)
$12=10 (step pulse, usec)
$13=0 (step port invert mask:00000000)
$14=192 (dir port invert mask:11000000)
$15=25 (step idle delay, msec)
$16=0.020 (junction deviation, mm)
$17=0.002 (arc tolerance, mm)
$19=0 (report inches, bool)
$20=1 (auto start, bool)
$21=0 (invert step enable, bool)
$22=0 (invert limit pins, bool)
$23=0 (soft limits, bool)
$24=0 (hard limits, bool)
$25=0 (homing cycle, bool)
$26=0 (homing dir invert mask:00000000)
$27=25.000 (homing feed, mm/min)
$28=500.000 (homing seek, mm/min)
$29=250 (homing debounce, msec)
$30=1.000 (homing pull-off, mm)
ok
Enviado: g0x50 
Recebido:
ok
Enviado: x50 
Recebido:
Enviado: x50 
Recebido:

closed 2023-06-19 by dougkusanagi