grbl/grbl #1070: Do we lose much if we break a straight move into multiple segments?

cspwcspw, 2016-08-28:
Suppose I'm at X0 Y0 and want code 
G01 X100

Is this significantly different on grbl from breaking the line into two fragments  (Apart from the negligible (for me) communication and parsing overhead) 
G01 X50
G01 X100

My algorithm for generating my g-code finds generating the second one logically easier.   But my quick experiments seemed to indicate grbl had no trouble: its motion planner made the move just as smoothly in the second case, even when I had very slow acceleration rates set.

For circular motion, would the same generally be true?  If I want to sweep through 300 degrees of arc, and I generate two G03 commands that each sweep 150 degrees, should I see any significant difference in actual motion? 

Thanks
Peter

jahnj0584, 2016-08-28:
Only thing I can think of is some speed lost from deceleration to prepare
for the next command. Not sure how much lookahead GRBL has.

chamnit, 2016-08-28:
@cspwcspw : The easiest way to find out is to test it, but yes, Grbl will smoothly transition between straight motions when it is able to. It's rare that it won't and that is limited by how much total distance is in the planner.

closed 2016-08-28 by chamnit

cspwcspw, 2016-08-28:
Thanks Chamnit.  Would this likely also be the case if feed rates changed between the two segments?  For example, at X0 Y0 if we had   
G01 X50 F1000   ; the material is thinner / softer / non-existent here, I can feed faster
G01 X100 F300   ; there is more to cut here, at a reduced speed ...

Should I expect to gently slow down before X50 using the acceleration parameters, so that by time I hit X50 I am more or less at the target speed?   (Or, when we drive our cars and see a reduced speed limit ahead, do we slow down so that by time we enter the slow zone we're already under the new speed limit?)

Thanks

jahnj0584, 2016-08-28:
You should always use a consistent feed for cut quality. Are you not using
depth of cuts? Not everything should be cut in one pass.

swarfer, 2016-08-29:
Hi Peter
The straight line motions will be smoothed out by the planner, no problem there.
However, if there is nothing to cut then rather use a G00 rapid motion.

For arcs .... it is generally advisable to use arcs of less than 180 degrees anyway, particularly if you are using the R format.    >=180 degree arcs can suffer from various math anomolies and get reversed.    The IJ format is better at avoiding this problem.    BTW this is not a GRBL specific thing, all controllers suffer from this problem and many major manufacturers recommend (or enforce) short arc segments to avoid problems.