grbl/grbl #1071: Enable M17/M18 to control power to steppers

ThePiGuy, 2016-08-28:
It would be great if these two commands (M17/M18) could be implemented in GRBL such that M17 enables power to the stepper motors, and M18 disables power to the stepper motors.

I know that current by using $1=255 one can enable/disable the steppers after a move, but a pure g-code solution would be excellent.  Possibly a configuration option to set the maximum time to keep steppers enabled (say 1 minute) could be introduced so that the steppers can't be left on (leading to them being worn out) for longer than the time it takes for someone to do a manual tool change.

chamnit, 2016-08-28:
@ThePiGuy : FWIW, I'm starting to think about adding more Marlin-style g-code commands into Grbl's parser. Even though the 3d printing community has just made up most of it to fit what they need it to do, it's becoming part of a new standard. I think it'd be wise to start incorporating some of those things when it makes sense to. The stepper enable/disable is one of those. 

For the 1 minute time-out request, this would require an additional timer, which Grbl does not have available. (Technically it does if one were to remove a compile-time option completely). In general, I don't think this is a good idea. Steppers typically lose position when they are enabled and disabled. Usually due to the micro stepping being reset every time this happens. For the most part, modern stepper motors and drivers are ok to leave on for extended periods of time.

However, I did add something similar to the GrblMega fork (for the Mega2560) for timeouts already. This was to detect for USB disconnects during a job.

langwadt, 2016-08-28:
if the motors or drivers get too hot while idling they'll get too hot running so it really shouldn't be needed. 
because I use drivers with SPI setup I just drop the motor current to a lower value instead of disabling

109JB, 2016-08-30:
As far as steppers go, I feel that way too many people get worried about their steppers because they are supposedly "hot".  Hot is relative, and the majority of steppers are rated for case temperatures of 100 deg. C, or 212 deg F.  Way to hot to touch with your skin, but not too hot for the motor.

As for disabling steppers, as was mentioned, in a microstepping mode, you will lose at least those microsteps from a full step when you disable and re-enable the driver. I know that the smaller drivers used with shields and such may not have this capability, but larger drivers have an idle current setting that doesn't disable the driver, but reduces the current when no steps have been commanded for a certain amount of time. Mine can be set to got to 40% current when idle, but I just leave mine set for full current all the time.

I have absolutely no problem in regard to adding Marlin commands, or any other commands as long as it is done in a good way.  Marlin is 3d printer specific and as such they have added a ton of commands, which is fine, but they have also changed existing commands. For example. M30, which is "Program stop/rewind" in RS274 is "Delete file from SD" in Marlin. Likewise, G10 which is used for work coordinates and tool offsets in RS274 becomes "Retract filament according to settings of M207" in Marlin. With all the numbers in the integer world, they chose to use M-code numbers that were already in use.  Why? I have no idea, but please don't do that. Please leave the core RS274 commands as they are, even if not implemented in GRBL yet/ever.

The 3D printer community already has Marlin and a host of others, many based on earlier versions of GRBL, that work well for them. GRBL itself is a great program and fills a niche for primarily Cartesian coordinate machines like milling machines, routers, pen plotters, etc.  It fills this niche rather nicely in my opinion. It has been extended to other uses, which again is fine, but my hope is that it stays at least mostly true to its roots as a Cartesian coordinate g-code interpreter. I also hope that it maintains a code set that is useful in production CAM software like it is now.

Thanks,

John B.

chamnit, 2016-08-30:
@109JB : Lol. Don't worry about Grbl getting too out of hand. There are some undefined functionality that hobby CNC machines need, like motor power/current. Grbl has been handling that with its own set of '$' commands, but as things will get more capable very soon, these '$' commands will need to get fleshed out even further. Technically some of them function already like Marlin's M codes, so it could be interpreted as Grbl being the one bastardizing the g-code standard.  

Anyhow, I've been just thinking that it'll be easier for people from the 3d-printing community to pick up later, more powerful versions of Grbl if it accepts similar commands. I'll have to really study it to see what makes sense and what doesn't.

Harvie, 2018-03-05:
Any news on this one? This would be SUPER useful for me. Especially for manual toolchange you could do
M17 M0 M18

So you will not move the axes while changing the mill bit and then you can unpause and continue. There are lots of machines using belt drives (eg. Shapeoko) or rack and pinion, that are easy to move by hand when motors are not enabled, so it's hard to loosen and tighten the tool collet without moving them.

This feature should be pretty easy to add...

chamnit, 2018-03-05:
Not as simple as you may think. The implementation isn’t hard and is actually available as a sleep command in v1.1. But it auto resets Grbl when woken up.  It works different for the following reasons. 

Stepper motors and drivers lose position upon a enable and disable. Some lose the micro step and motors will drift without any power. So for a tool change this makes it completely useless.

In addition, you need to measure the z offset when you do a tool change. It can be slightly or completely different from the tool before it. This would result in a bad crash. 

Certainly in some situations and advanced CNC users, this may not be a problem. But for the mass majority, who are new to CNC, it would be an unending problem of complaints, questions, and headaches. 

Finally M17 and M19 is a construct of the 3D printing world. It’s not used in subtractive machining and the ability to disable motors is usually limited by the manufacturers in varying degrees for the same reasons above.

closed 2018-03-05 by chamnit

Harvie, 2018-03-05:
@chamnit 

> Stepper motors and drivers lose position upon a enable and disable. Some lose the micro step and motors will drift without any power. So for a tool change this makes it completely useless.

No they don't loose anything. They only lose in case you disable them. All this is for disabling of disabling them. Especialy made so you don't lose anything.

> In addition, you need to measure the z offset when you do a tool change. It can be slightly or completely different from the tool before it. This would result in a bad crash.

I am fully aware. There are lots of crash dangers already in CNC machining and users have to learn how to handle it. This is just one of the cases. This has 3 cheap and obvious solutions:

1.) Use tools with plastic stoppers (or put on your own stoppers with setscrew). BAM. problem solved. perfect for beginners:

[image]

2.) people use simple fixtures. simple wooden board with hole may be enough to make sure tool is extended just the right amount from spindle.

3.) automatic tool probing. (this is kinda hi-tech, but can be done).

> But for the mass majority, who are new to CNC, it would be an unending problem of complaints, questions, and headaches.

Why would somebody who is new to CNC start creating g-code containing toolchange? I guess beginner would start with simple, single tool, g-code.

chamnit, 2018-03-05:
Ok. I reread the original post. It’s best not to have the steppers disable at all. I’ve come to regret the step idle delay setting altogether.  As most modern stepper drivers produce very little heat and can be left on just about indefinitely. 

If you do need to disable them, it’s assumed to be at the end of a job, which the v1.1 sleep mode command is designed to handle.  Ideally, there should be an auto disable after X minutes, rather than after a few ms that Grbl does. 

Yes. Of course there are solutions to tool changing, but you’re assuming that newbies will know everything about it and have no questions or problems.  Not true. Totally not true. Even if you spell everything out verbatim, there is always a portion of the user pool that will never read directions and get everything backwards. I see this almost daily. 

Tool changes are easy when a GUI is designed to implement them. No need to dealing with special gcode for disabling motors. For example, Carbide3d’s Nomad is built with one. They insert all of the tool change commands for you and automatically probe the tool length offsets. It works very well and there is very little room for error and misuse. I know that bCNC (and maybe UGS too) has a way for users to install custom routines for their machines as well. 

I’ve found it’s best to leave setting up tool change stuff to the OEMs or to advanced users with the know how. It’s impossible for a firmware to account for every situation or every machine type or every workflow preference.

mpatoulachik, 2018-03-05:
I agree that you need to keep the steppers on if you don't want to lose steps, but in some cases it is convenient to be able to disable the steppers to manually move the axis (handwheels behind the steppers, or like with a laser cutter, just pushing the head around). 
Or the opposite, when doing some stuff on the machine, it is nice to force the steppers on so nothing moves by itself. 
This means you have to re-home after that but you don't lose anything else.
So I believe some people would have a great use of some kind of command to activate/deactivate the steppers (GCode or Grbl).

chamnit, 2018-03-05:
@mpatoulachik : Yes. Agreed. You can accomplish this with the new sleep command in v1.1. It institutes an auto-reset, where it forces you to acknowledge the potential lost position and re-home.

langwadt, 2018-03-05:
@mpatoulachik  just beware that driving the steppers by hand will force current into the supply  and possibly damage the electronics

109JB, 2018-03-06:
How about a simple switch supplying 5V to the enable pin of the driver.  Not everything has to be complicated or software driven.

109JB, 2018-03-08:
Actually a simple dial to adjust the feed override would be fantastic. 

I was simply making a suggestion because many times people get too focused on one potential solution and ignore all others. The forest for the trees scenario.

Harvie, 2018-03-08:
@chamnit you should not regret automatic disabling of motors. You saved my drivers several times when i was new to thing and didn't know that i should not unplug motors while drivers are under power. nothing bad happened, because they were automaticaly disabled by grbl. Also i still haven't finished all wiring stuff and the motor wires came loose on me several times, which would burn the drivers as well. Luckily in few days proper connectors with locks should arrive, so it won't be able to pull my plugs out anymore.

@109JB I meant something like this:
[image]

However when you started this HW solution to SW problem thing... I wonder if i can use this pin as input:

[image]

But i guess that would only fry the output of the atmega if i tried to short these two pins or something.

However obvious way to disable motors using single switch would be to simply disconnect the 24V power supply. I am planning to install such switch.

109JB, 2018-03-08:
@Harvie You could probably use those pins on the shield by simply bending or cutting off the pin that plugs into pin 8 of the arduino. That would isolate the Arduino and then you could add a switch like this.
[image]

Harvie, 2018-03-08:
@109JB or just patch GRBL to set EN pin as input (possibly with internal pullup), so you don't destroy anything by applying 5V to it.

109JB, 2018-03-08:
@Harvie - Sure, but since it has already been stated that there will not be an official mod to the Grbl firmware for this, the bending of pin 8 and adding a switch is an option that would work for the vast majority of cases where disabling the steppers is desired. It was suggested as a work around since it appears the only way you are going to get what you are asking for is to mod the Grbl firmware yourself. If that is what you want to do then nobody is stopping you.

Also, my post about using a simple switch was in response to @mpatoulachik post where he said:

> I agree that you need to keep the steppers on if you don't want to lose steps, but in some cases it is convenient to be able to disable the steppers to manually move the axis (handwheels behind the steppers, or like with a laser cutter, just pushing the head around).
Or the opposite, when doing some stuff on the machine, it is nice to force the steppers on so nothing moves by itself.
This means you have to re-home after that but you don't lose anything else.
So I believe some people would have a great use of some kind of command to activate/deactivate the steppers (GCode or Grbl).

Just about everything he mentioned in that post could be alleviated by a simple switch to enable or disable the drives.

Going back to your original post in which you said 

> Especially for manual toolchange you could do M17 M0 M18

The M17 would disable the drivers and immediately you would lose position at least a little. When a stepper driver is disabled and then re-enabled it re-enables to a full step position whereas the position before disabling could be a microstep position, so you could lose a position by 1/2 step just from the operation of the drivers.  Add to that the possibility that the motor is free to rotate while disabled and any rotation before re-enabling will also add to the lost position.

I personally see absolutely no reason to disable the drivers for a tool change. I do tool changes all the time using Grbl and here is how I do it. Grbl has the ability to use G28 and G30, All you have to do is define either G28 or G30 as a tool change position, and then program

G28
M0

The machine will move to the position you define for G28, and then pause for your tool change while keeping the motors enabled.

I'm not trying to be argumentative. I'm just trying to present options for those that can use them. You need to remember that many more people read these issues posts that those that actually post in them. While you may not like my suggestions, they may be beneficial to someone else.