grbl/grbl #1072: grbl Nc switch

Encarnacao, 2016-08-29:
i wanted to put NC switches on grbl9 but im confused in what to do

i nedd to enable //#define DISABLE_LIMIT_PIN_PULL_UP in grbk config
i need to invert limit pins on grbl  $5=1
and i need to put a external pull down like 1k to gnd and the end of pin conected to +5v?

correct me if im wrong in anything
all what im confused now is how to wire the pull down could u show me any schematic with the  correct wiring to dont mess up?

thanks in advanced

chamnit, 2016-08-30:
@Encarnacao : The Grbl invert limit pins setting $5 sets if a pin high or low means triggered. So, it'll set this for NC or NO switches. You don't need to use the DISABLE_LIMIT_PIN_PULL_UP option. It's for special circumstances. Just connect the limit pin to your switch to ground. Add an optional small resistor between ground and the switch to limit the current. Add a small capacitor between the pin and ground if you have noise issues.

closed 2016-08-30 by chamnit

Encarnacao, 2016-08-30:
the problem im facing is machine is working and then after 20 min or so it stops and only dremel keeps working i had the limit switches with 1uf capcitor betwen ground and also switch  for the optional buttons stop, pause,resume,emergency also with 1uf capacitor when i removed the switch conections the machine start to last longer but still keeped stoping so i removed also the buttons and then i tested de machine for 2 hour and it never stoped, so that makes me belive that im facing eletric noise issues, sense the internal pull ups of arduino is like to 20k that makes them a litle weker so i tought that having like external pull downs with like 1k resitor would make them more stronger

gerritv, 2016-08-30:
You need to read up on debouncing, it is not as simple as just adding a cap although that might seem to solve some small issues. See my (gerritv) post in https://github.com/grbl/grbl/issues/542 on Feb 5. The whole thread is worth a read.
Without an oscilloscope though you are just guessing at what is happening. E.g. are your leads shielded, and the shield grounded at only 1 end? etc. Noise is a complex issue to address. Spindle motors with brushes are the worst culprit.

Encarnacao, 2016-08-30:
i have made some small changes now i have put all shielded cables to reduce noises, but still think that having NC switch would probably be a better solution sense if something happen like a broken wire or switch would stop the machine.
i have cross before at that post is a long discution, a litle confusing tought sense english not my first language, would be great to have some type of shematic with instructions in how to wire correctly a NC switch using external pull ups or pull downs if you removed the DISABLE_LIMIT_PIN_PULL_UP option.

gerritv, 2016-08-30:
This thread seems to describe what you are looking for, with some diagrams. http://www.shapeoko.com/forum/viewtopic.php?f=3&t=5156&p=37798#p37767