grbl/grbl #1074: Hard Limit Differentiation?

cspwcspw, 2016-08-31:
I'm getting some false limits triggered, as many have before me!  I do have shielded cables, and the problem has only occurred when the spindle switches on, but I will have to install the mods for the high-filter bypass.

But I don't share X,Y,Z limits: they are on different pins.  So it might be useful to know "Which axis is my culprit?" My Z-switch limit cables run quite close to the spindle, and mains power and those limit signals travel in the same drag-chain, so those are my chief suspects.  

I see in grbl code in limits.c there is a function **uint8_t limits_get_state()** that returns the info I'd like to see.  Would it be difficult to echo this back to the GUI as part of the hard limit alarm or the alarm status?   (A bit like soft error / parser messages that carry return error codes with them.)   Or perhaps store it somewhere (e.g. have an extra hash parameter) that I can query once I've cleared the alarm. 

Thanks.

jahnj0584, 2016-08-31:
Check out the UGS git hub. FaulterLee made me an arduino program that told
you when a limit switch was activated. It flashes over GRBL, so you'd have
to manually turn on the spindle and see if it causes it.

lalo-uy, 2016-08-31:
Don´t worry with which one is the culprit. Put filters on all 3 lines.
May be Z is the problematic now, but the others may truble in the future.

chamnit, 2016-08-31:
@lalo-uy : Good suggestion. I can see this being very useful. I'll see what I can do in terms of cramming it into v1.0.

closed 2016-08-31 by chamnit

chamnit, 2016-09-03:
@lalo-uy : Looked into this today. I don't think adding which limit threw a hard limit would work dependably. If you have electrical noise in your system, it can trigger the interrupt to set a hard limit alarm, but not detect which axis switch did when the processor gets to executing that part of the code. Plus, Grbl already can report the limit switch state inside of the status reports when the $10 enables it.

That said, I can add a limit_state_check() and provide a feedback message that a limit switch is still engaged wherever it makes sense to, like after the welcome string and an alarm unlock.

cspwcspw, 2016-09-04:
@chamnit:  Thanks for looking into this!  

Yes, discovering the limit switch states are provided for nicely via polling, and the suggestion that you might report hard limits that are still engaged after a reset might be helpful too. 
 
But I was really asking if we could find out what pin caused the transient triggers,  In general, noise is not going to be "poll-able".  So if the interrupt service routine (ISR) doesn't know or can't find out which pin gave rise to its execution, or if multiple limit pins are sharing the same ISR, it sounds like extra work for perhaps minimal returns.

chamnit, 2016-09-04:
@cspwcspw : I understood what you meant by knowing which pin is noisy and triggering the hard limit interrupt. By having this, it would provide a really nice tool for users when dealing with EMI. 

Just to be sure my memory on how the pin change interrupts work, I checked the data sheets late last night. The AVR pin interrupt detects a pin change on a port (8 pin groups). It does not retain any information on which of the pins in a port group that threw the interrupt flag. As far as I understand it, you have to poll the port within the interrupt to get its state. So, if you have a transient noise problem, detection by polling will not be guaranteed. 

In addition, the 328p only has 3 pin change interrupt routines. It's possible to move the limit pins to different ports (D0-D7, D8-D13, A0-A7) to isolate axes to each port. However, it would require re-writing lots of the code to accommodate this; pins would need to be swapped around; and you'd lose the input control pins (reset, feed hold, resume) since they also use the pin change interrupt. So, I'd say its extra work for minimal return.

cspwcspw, 2016-09-04:
@chamnit:   Thanks, that makes perfect sense. I didn't realize pins were aggregated into ports.  I haven't got close to reading the code in any detail, let alone the data sheets, and its not my primary focus anyway.  So it's great to have the support and insight that you provide.

gerritv, 2016-09-04:
A flipflop with 7417 in front of it should provide a latch to show which pin triggered? Somewhere I have a digital tester that has this function built in.
A logic analyzer, even a cheap one, would also do the job of capturing.

cspwcspw, 2016-09-05:
@gerritv: Nice idea.  I'm not an EE, but to me it seems that the internal pull-ups from the Ardunio are somewhat weak, so they are sensitive to noise.  So a filter with a capacitor (and perhaps a resistor?) can compensate for that,  Or one could abandon the Ardunio internal pull-ups and provide your own.  So if I were to put anything in front of the Ardunio pins, I'd probably try to arrange a slightly stronger pull-up current to the switches, and I suspect the problems would likely go away.

Some folk report success if one just inverts the logic (and grbl has configurable settings to accommodate this).  So if the switch is normally closed the lines will carry a "strong" ground, making it difficult for stray electrical noise to pull the voltage up.   Multiple switches tied to the same limit pin must be rewired in series rather than in parallel,