grbl/grbl #1075: Baudrate changes for implementing RS232 serial conversion

MeJasonT, 2016-09-01:
Hi,
Ive been trying to change the baud rate in config.h file so that i can slow down the 115200 to 57600 baudrate to help in connecting grbl via a serial link instead of USB.  The unfortunate side effect of USB is it is incredibly noisy and unreliable.  My current project involves running a CNC Plasma table with a cheap Chinese plasma cutter, its as noisy as hell to start with (emf and hf noise).  It appears that the data rate is too fast for the Prolific USB to RS232 converter to handle and Grbl will only run for 2 mins before crashing (just talking to GRBL and receiving verbose commands not running the machine).  I can only find the baud mentioned once in the code, odd thing is if i comment it out and upload the code to my Uno/Grbl board it still runs at 115200 baud.  Where is the white-mans magic, why is this even possible.  I used the Arduino EEprom eraser code and blanked the memory then uploaded Grbl with a 9600 baud setting - it still sows and connects at 115200.  I have deleted all GRBL files from my PC and started with the recent build (coppied grbl into library folder etc).  still cant affect any changes to baud rate.  Please can you tell me what im missing.
I have it on good authority that using a serial cable connection to my machine will stop the crashes i suffer on the USB link to the machine.  The hf frequencies doled out by the plasma cutter when cutting play havoc on the USB, ive used opto isolators and filters etc but the usb  keeps being an issue.
I would like to say thank you for sharing this fantastic project and firmware, its pure genius.

gerritv, 2016-09-01:
Config.h, around line 37 is the define for BAUD_RATE. Change the value there and recompile. The UNO will still require 115200 to program but after starting up Grbl will run at the baud rate you chose.

chamnit, 2016-09-01:
Make sure you are editing the file in the Arduino library folder, not your download folder. Also, Chinese USB-serial converters are known to fail during stream, when the stream has a lot of data traffic. Use Arduinos with an FTDI chip or an atmega8U/16U. Not the CH380s.

MeJasonT, 2016-09-01:
Thanks gerritv and chamnit and cheers for the heads up on the CH380 chip/driver that's exactly whats in the prolific USB converter and probably the answer to the serial comms crashes i'm experiencing.
i will have to use a Uno as suggested and then convert ttl to RS232 using a Max232 ic/pcb.  The link will have to be RS232 not just TTL serial and then same on the other end with another converter IC.  For your info the plasma table works well and iv'e made a few cuts with it, i'm now fine tuning and trying to find a way of getting the mechanical limit switches back working as they are also crashing GRBL/USB Comms due to hf noise.  Adding Earth wires and shielding to everything plus the addition of ferrite cores got me from no functionality to 90% there. Project is posted here [http://www.homemadetools.net/forum/cnc-plasma-table-using-cut50-cheapo-plasma-cutter-48452#post69137](url)
With the creative acumen and good will of you two guys i have also managed to convert my old milling machine into a CNC as well.  Cant find anything to make now, i had loads of ideas when i wanted the technology.  I guess i will have to build another CNC machine, perhaps convert my Lathe.

MeJasonT, 2016-09-01:
Hi Gerritv,
"changing Config.h, around line 37 is the define for BAUD_RATE" has no affect"
actually if i comment it out and upload it still uploads without error and keeps running at 115200 baud.
im very confused by this. having deleted all grbl folders from my pc and unpacked a new version in the new grbl ide library folder.  im very confused - there is magic in that there code.

chamnit, 2016-09-01:
@MeJasonT : I'm considering this matter closed since this isn't a problem with Grbl, but a USB-serial converter. With baud rate, you likely have missed where the Arduino library puts user-added libraries. It's often in the Documents folder.

closed 2016-09-01 by chamnit