grbl/grbl #1077: Path Planner fails: Instant change in velocity.  Acceleration ignored.  Causes Z to drop!

3dDriver, 2016-09-02:
The Z is not obeying the deacceleration/acceleration when moving DOWN, then UP.
Instead, it is accelerating down, to a final velocity, V, then *instantly* moving UP at V.
This causes Z to drop due to missteps (even at low accelerations).
Bug is repeatable 100% of the time for me.

I have been using GRBL for 100s of hours with no problems.  This bug popped up in new particular file.  I narrowed down the file to a few lines to recreate it easily.
If you use the machine settings, below, $122 = 0.4 mm/sec^2, you can actually see/ hear the acceleration ramp is incorrect.  Or you can change it to $122 = 75, which is what I normally run with the CNC, and see the machine lose steps.  In either case, velocity does not ramp down as Z approches the bottom.  This causes problems with lower Z accelerations: even 50mm/s^2.  

I have narrowed down the bug to a few lines of code:

 "cws HERE1" - this Z-2.5 causes instant Z velocity change, at Z0.0, below.
 If you change from Z-2.5 to Z-3.5, then proper acceleration on Z axis occurs.

"cws HERE2" - instant UP velocity occurs here! 

G90
G21
F550.0
G1 Z0.000
G1 X74.925 Y159.603 Z-2.248
   ( cws HERE1 )
G1 X74.934 Y159.553 Z-2.5 
G1 X74.925 Y159.603 Z-2.248
   ( cws HERE2 )
G1 Z0.0

 $0	10	(step pulse, usec)
 $1	30	(step idle delay, msec)
 $2	0	(step port invert mask:00000000)
 $3	6	(dir port invert mask:00000110)
 $4	0	(step enable invert, bool)
 $5	0	(limit pins invert, bool)
 $6	0	(probe pin invert, bool)
 $10	3	(status report mask:00000011)
 $11	0.020	(junction deviation, mm)
 $12	0.002	(arc tolerance, mm)
 $13	0	(report inches, bool)
 $14	1	(auto start, bool)
 $20	0	(soft limits, bool)
 $21	1	(hard limits, bool)
 $22	1	(homing cycle, bool)
 $23	11	(homing dir invert mask:00001011)
 $24	225.000	(homing feed, mm/min)
 $25	700.000	(homing seek, mm/min)
 $26	250	(homing debounce, msec)
 $27	1.000	(homing pull-off, mm)
 $100	266.666	(x, step/mm)
 $101	266.666	(y, step/mm)
 $102	640.000	(z, step/mm)
 $110	700.000	(x max rate, mm/min)
 $111	1000.000	(y max rate, mm/min)
 $112	800.000	(z max rate, mm/min)
 $120	100.000	(x accel, mm/sec^2)
 $121	100.025	(y accel, mm/sec^2)
 $122	0.200	(z accel, mm/sec^2)
 $130	540.000	(x max travel, mm)
 $131	542.000	(y max travel, mm)
 $132	102.500	(z max travel, mm)

Chambit, thanks for all your efforts. GRBL is great!  I hope this sheds some light on some users Z problems / missteps.

Chris

chamnit, 2016-09-02:
@3dDriver : That's unusual, but I'll take a look at it this weekend. One thing that jumps out at me is that you have a very low z-acceleration and the motions are very small (tens of microns). Typically this type of issue is related to the z-axis settings or stepper driver, but from what it sounds like, you have tested this. If you can, try reducing the junction deviation setting to something like 0.01 or 0.005. 

There is a known potential issue losing steps that has to do with very shallow z-axis motions when doing a helical plunge and retract. It's not usually a problem, but typically occurs when you have a very different z-axis acceleration setting compare to x and y. I've updated the planner in the grblMega project to help improve performance through motions like this. So, far I've heard it works better. I'll likely pull it into Grbl v1.0.

chamnit, 2016-09-02:
@3dDriver : Also try using the most recent version of Grbl v0.9j. I think this might be an old bug in an earlier version of v0.9 that has already been resolved.

3dDriver, 2016-09-02:
I'll try it tomorrow... thanks!I had read about the helical bug, but I don't think it is the same bug.

chamnit, 2016-09-02:
@3dDriver : Had a chance to step into the "lab" tonight. I can confirm something weird occurring. Definitely a instant velocity change, but not quite what you've described. I'm also using the grblMega branch on a 2560. So, I get instead a stop prior to the z-2.5, around z-2.48. Then it continues to -2.5 and does and instant direction change toward Z0.0. I didn't notice any second instant velocity occurrence. Can you verify if this is the similar to what you are seeing?

3dDriver, 2016-09-02:
On your setup, Does the z slow down to 0 velocity when going to reach Z-2.48?  
Mine did not slow at all.  It accelerated down the entire time to reach V near Z-2.48 or Z-2.5 (I can't tell exactly...I'm going by pure visuals), then started upward at V.
I could only observe via the reduced z mm/s^2. (No scope etc).
I can't comment for Mega.  I used Uno.
Btw - The only reason I used reduced Z mm/s^2, was to observe behavior in slow motion.  I believe this is what also happens at Z= 75mm/s^2.  At that acceleration, V is higher and my CNC can't handle the instant changed V (actually 2V).

3dDriver, 2016-09-02:
One thing that I mentioned might be subtle but is noteworthy:

 "cws HERE1" - this Z-2.5 causes instant Z velocity change, at Z0.0, below.
If you change the "Z-2.5" to "Z-3.5", then proper deacceleration / acceleration on the Z axis occurs
   "

Note: changing the value mentioned, affects the command that is 2 lines down from the change.  It's like the path planner, at the z0 line, has a bad calculation/memory of its previous velocities/coordinates from two lines above.  The change "fixes" the calc.

chamnit, 2016-09-02:
@3dDriver : I'm having a hard time understanding exactly what you're saying, but I've gleaned that its different from what I've seen. I'll setup my test rig for an Uno and Grbl v0.9j (current) and retest. 

FWIW, I don't think it's the planner itself. Early debugging check show its operating correctly. It's more likely it's a numerical issue with these very small movements in the step generator, because it's the one that generates the acceleration profiles.

chamnit, 2016-09-02:
@3dDriver : Well, this is an interesting result. I get no such issue on an Arduino Uno with Grbl v0.9j running on it. Your program with your settings runs just fine. No instant velocity changes. Can you update to v0.9j and check your system?

3dDriver, 2016-09-02:
Ok.  But first I changed "junction deviation" to .005.  It did not fix the problem when z accel was 75mm/s^2.  It did work with z accel = .2.

Next, I upgraded.  The simple test case, above, works.  

I'm running a job in the air/no cut to see if any steps go missing...so far so good....

I took a video of the cnc accelerating / instant velocity change for the original bug.  If you're interested, I can email it to you if I can find your email.

chamnit, 2016-09-02:
@3dDriver : You can email me at gnea.research at gmail dot com. This is an email address that I use specifically for Grbl stuff.

I was just looking at the commit logs. Grbl v0.9g had a bug when there are multi-axis moves that returned along the same path. It was fixed on 2015-02-10.

Not exactly sure what's going on with the Mega version, but since this seems to be resolved, I'll close this issue. I'll reopen it, if I hear otherwise.

closed 2016-09-02 by chamnit

3dDriver, 2016-09-02:
After 3 air runs, and an actual cutting run.  All is good! Thanks for your efforts, Chamnit!

chamnit, 2016-09-02:
@3dDriver : Thanks for letting me know. Also, there is nothing wrong with the grblMega fork. My test rig was miswired.

109JB, 2016-09-03:
So what was the fix?  Upgrading to 0.9J??

chamnit, 2016-09-03:
@109JB : Yes. It was fixed about a year and half ago. Just update to the current version.

109JB, 2016-09-03:
@chamnit : No problem. I already had the latest version. I just wasn't following what the exact fix was since there was discussion about accel's, junction deviation, etc.