grbl/grbl #1078: Bad UBRR calculation?

coco243, 2016-09-04:
I don't think that the UBRR formula is correct in the serial.c

uint16_t UBRR0_value = ((F_CPU / (8L * BAUD_RATE)) - 1)/2 ;

it shouldn't be?

uint16_t UBRR0_value = ((F_CPU / (16L * BAUD_RATE)) - 1) ;

gerritv, 2016-09-04:
The code is/has been working fine for years with those settings. What makes you think it is wrong?
A very very long discussion in baud rates didn't raise this either https://github.com/grbl/grbl/issues/837

langwadt, 2016-09-04:
the second equation is the one show in the datasheet, as far as I can tell it comes down to rounding vs. truncating  

((16M / (8 * 115200)) - 1)/2 = 8.18
((16M / (16 * 115200)) - 1)  = 7.68

if you round they both result in the correct 8 ( 111111 baud)
if you truncate the second equation is 7 (125000 baud)

closed 2017-01-26 by chamnit