grbl/grbl #1080: Noise and limit switches

tyfr12203, 2016-09-10:
My limit switches seem to be extremely sensitive to electrical noise. I had everything set up on my cnc laser engraver, all wires run through the cable chain and plugged in to my arduino with a shield. Soon as I try to do a manual move I get an immediate hard limit fault.If I unplug the limit switches everything moves fine. It got to the point where I had to take all of the wiring for the limit switches out of the chain and hold them up in the air away from everything to get it to function normally, which it did. So I'm assuming it has to be electrical noise but what can I do about this? Thanks

gerritv, 2016-09-10:
I would  search the issues here, Noise is mentioned at least 20+times in various issues.

PicEngraver, 2016-09-10:
Add 0.1uf CERAMIC capacitors between each limit switch pin input and ground at the Arduino.  Several other suppression circuits are in use by others, but this may be all you need.
John Champlain
www.picengrave.com

tklus, 2016-09-10:
I am also having issues with my machine and the limit switches being very sensitive. I am running shielded stepper cables, ran the limit switch wires on the outside of my energy chain and tried to hook in capacitors but after I hooked up the capacitors, I couldn't get the switches to trigger at all.

Does someone have a wiring diagram that shows how to wire in the caps? It would be cool if we could come up with a little circuit board we could scratch build for wiring in the caps.

Thanks
Tim

PicEngraver, 2016-09-10:
Maybe this helps?
http://www.picengrave.com/forum/viewtopic.php?f=13&t=155&start=230
John

tklus, 2016-09-10:
Thanks John! I will give that a shot. I had the cap between the ground and limit pin... That didn't seem to work :)

DOUG888, 2016-09-11:
I developed this board to solve interference problems, it has the added advantage of converting the pins to suit the later versions of the CNC SHIELD V3.0 to the later version with PWM enabled. I have been using this for the last 2 weeks and all seams ok, but as always...use at your own risk.
This is a Design Spark circuit, picture and pattern attached, hope it helps
[CNC SHIELD V3.51 OFFSET with RESET.zip](https://github.com/grbl/grbl/files/465695/CNC.SHIELD.V3.51.OFFSET.with.RESET.zip)
[CNC SHIELD V3.0 offset.zip](https://github.com/grbl/grbl/files/465696/CNC.SHIELD.V3.0.offset.zip)

 this helps.