grbl/grbl #1081: Homing fail after changing $100 - $102

MeisterMax, 2016-09-14:
Hey there,
I've a weird little problem configuring my cnc machine. I have limit switches installed on all axes, homing and hard limit are enabled. Everything, including homing worked just fine as I tested it. Then, however, I changed the step/mm settings for all axes from the default values (it was something like 250, wasn't it?) to the ones fitting my machine, which is $100 = $ 101 = 40.000 and $102 = 47.244. As I tried homing afterwards, the z axis moved upwards, then down a bit, just as it should, and then I got an ALARM: homing fail, without the x and y axes having done anything at all. I've installed 0.47uF capacitors in series with the NO limit switches by the way, so this shouldn't be an interference-problem. Anyway, as I changed $102 to something higher like 180, and performed a homing cycle, the z axis got into it's home position, x and y started moving, and i got the very same problem, just that it was with the x or y axis this time. Has anyone _any_ idea what to do? I'd greatly appreciate your help.
Max

chamnit, 2016-09-14:
@MeisterMax : There are only a few conditions where Grbl throws a homing fail error. IF it doesn't find a limit switch within the search region (~1.5x max travel). If it doesn't clear the limit switch when pulling off of it, which is tied to your homing pull-off setting. Or if you soft-reset Grbl. Electrical noise can cause the pull-off to not recognize the machine has pulled away, but I think you just need to increase your pull-off value and check the switches to see if they are not physically sticky.

MeisterMax, 2016-09-14:
Ok, I'm going to try this. Thanks for your quick reply!

MeisterMax, 2016-09-14:
That's it! I just set the pull-off value to 3mm and everything worked. Thanks a lot. Great software!
Max

closed 2016-09-14 by MeisterMax