grbl/grbl #1082: How to find specific old versions in souce code (not binary builds)

token47, 2016-09-14:
Hello

I would like to download specific versions in source code from the repository. The same specific pulls that were used to compile the binary builds.

I can't seem to find it anywhere. The README only offer binary downloads, the branches and releases don't help either.

Looking through the commit logs and guessing is too much error prone (done that but there are cases where later the release date is changed and you don't know exactly which commits of that day goes into release, etc.).

Is there a list of releases in source form?

Thank you

chamnit, 2016-09-14:
@token47 : You can go through the commit history and get any source from any date you'd like. Generally, I highly advise against using older versions of Grbl, as there are likely numerous bug fixes. This is the main reason I don't manage minor releases, only major versions. It helps prevent having to dedicate time to debugging old issues that have been previously fixed in later versions.

closed 2016-09-14 by chamnit

token47, 2016-09-14:
Hi @chamnit!

Thank you for your response. Yes, indeed it's best to use the last version, I tottaly agree. And I can for sure download any specific commit. But the descriptions of the commits themselves do not specify for sure which of them were used to generate the builds.

In this case, I don't want to use them, I wan't to read them and identify changes between whole versions (and not by each commit) and also diff them against some forked versions found on the internet where the author inly provides the final version and not the diffs (such as grbl-servo, which is on github but was commited already changed on first versions instead of being forked from grbl original).

I just can't find where this boundaries are. I suggest you publish the specific commits that were used to make each build, perhaps making official source releases and linking them on the readme together with the binary downloads. Anyway, just pointing me to the commits now would be of great help. Specifically the "0.9i" release is the one I'm most interested now.

Thank you

The problem is that I can't find any correlation

chamnit, 2016-09-14:
@token47 : I really don't have the time for that. Sorry. You'll have to do some the work yourself and browse the commit logs. I do my best to make them as clear as possible what changes have been made to the code.

token47, 2016-09-14:
@chamnit ok, I understand. Thank you anyway.