grbl/grbl #1083: Problem with cnc shield

jszewczyk, 2016-09-17:
Hey.

I use Arduino Uno with cnc shield and DRV8825. I connected motors. I have problem with motors. Axis X doesn't work. Y works well. Z axis works only in one direction. I tried to change cnc shield or DRV8825 to another, but it works the same. Have you had any ideas for it? 

Thanks. Jacek

jszewczyk, 2016-09-17:
cnc shield 3.51, Arduino Uno R3

Encarnacao, 2016-09-17:
did you adjust the trim pot for the vref  for the motors that you have?
i would remove the  Y driver that works good and move to the ones that dont work first to see if all ok
make sure dont turn shield on  with out the motors to the drv8825, with out the load you can damage them

jszewczyk, 2016-09-17:
Yes I adjust vref. When I changed the motor and drv8825, problem appear on him. I see that problem is bind with a port.

Protoneer, 2016-09-17:
Its likelly the direction pin. Please check your solder joints.